mooltipass/minible #310: Query the time whenever TOTP is requested

bertvandepoel, 2021-11-30:
Currently, there are cases when the miniBLE hasn't synced yet by the time a TOTP response is requested. This results in login errors. The temporary fix is to restart moolticute. As discussed on IRC, there are several solutions, but the easiest would be if the miniBLE requested the time from moolticute (or e.g. the future android companion) every time a TOTP response is requested.

Jan-NiklasB, 2021-12-17:
I also have much trouble getting correct TOTP's from the miniBLE.
The miniBLE is always on (or in stndby) cennected to my PC, but the clock seems to have such a massive drift that after 3 days the TOTP doesn't work.
Also there should be an option to force the miniBLE (doens't realy matter if on the minible or in the app) to get the current time.

Please fix that in near time, it's an awesome feature but realy annoying if it doesn't works.

bertvandepoel, 2021-12-17:
Just so you know, @Jan-NiklasB, a good workaround in the mean time is to select "restart daemon" in the tray menu item of moolticute. That forces a clock sync and resolves the issue at that time. I'm quite sure limpkin has this issue earmarked as quite relevant for firmware updates, so hopefully this will be fixed permanently in the near future.

Jan-NiklasB, 2021-12-17:
Ah, good to know, thanks for the hint.
Yeah I guess so, also the support on my order was incredible fast, so I guess this is also a project of heart for him.

rastagraffix, 2022-01-16:
Another option would be to poll the Bluetooth Time service on the paired device.

limpkin, 2022-01-16:
@rastagraffix do you mean the host? I'm fairly sure the host doesn't expose such a service :)

rastagraffix, 2022-01-16:
All my Apple devices support the standard Time service, although they don't advertise it or allow connections to it over BLE unless you're a paired and connected device.  I'm currently working out the logistics of this on the Watchy; it's a much lower power alternative to the current SNTP over Wifi scheme they use.

rastagraffix, 2022-01-16:
To0 be clearer, not every device you connect to will support the Time service.  But if you're already connected to a device to do TOTP, it doesn't cost anything to see if GATT advertises the standard Time service and, if it does, connect to that and sync the time.

In theory this should work if you're connected to any iOS, iPadOS, or MacOS device.

limpkin, 2022-01-16:
oh that's quite interesting!

limpkin, 2022-04-03:
@bertvandepoel I'm currently looking into this one but I can't verify your initial statement:

> Currently, there are cases when the miniBLE hasn't synced yet by the time a TOTP response is requested.

On my setup, whenever a Mini BLE is connected, Moolticute will send the local time. Moreover, MC will regularly send the time to the Mini BLE so it can calibrate itself :)

bertvandepoel, 2022-04-03:
Oh weird, I thought you had already fixed this! A while ago this problem just went away for me and I've never seen it since. So some firmware or moolticute change seems to have rectified it, I think. But I'm not sure what exactly was the cause for the problem nor the fix.

limpkin, 2022-04-03:
oh neat! should i close the issue then? :)

bertvandepoel, 2022-04-03:
Let's maybe see if @Jan-NiklasB responds? Maybe he still has the issue. But if he doesn't (or doesn't respond), I would indeed suggest we close it.

Jan-NiklasB, 2022-04-03:
Hey,
to be honest I have to check this.
I don't really use the minible for TOTP since its easier for me to use andotp on my phone than scrolling through the whole list of credentials to get the totp....

Maybe if totp entering is part of the auto login it would become more relevant to me. ..

limpkin, 2022-04-03:
@Jan-NiklasB we've recently added firmware features to make the TOTP experience way better... stay tuned for bundle v9 release :)

Jan-NiklasB, 2022-04-03:
Now you have my attention 😂

limpkin, 2022-11-05:
hello everyone!
bundle v12 is available at beta-updates.themooltipass.com . Could you update your devices and let me know if this is still an issue? :)