mooltipass/minible #311: Five logins for same website not prompting.   "Login Confirmation Prompt" behavior...

CubeCentral, 2021-12-04:
### Expected behavior
When visiting a website for which one has several logins (_more than two?_), be prompted on the device with a list to choose from.
When the "Login Confirmation Prompt" setting is "Off", and one visits a website for which one only has a **single** credential, it should automatically send credentials _without_ prompting.  This setting should not change the behavior of presenting a list for a site with more than one credential.
If there exists more than one entry for a website, _always_ display the list of choices and wait for input.   The "Login Confirmation Prompt" setting should **not** change this behavior.

### Actual behavior
Today I went to a website for which I have five logins. When I visited this site, it would quickly automatically enter credentials for one of them without any input from me.  It would always choose the same one.  (_How is this saved?_)  I then turned the "Login Confirmation Prompt" setting to "On" and tried again.  This time it paused and asked me if I wished to log in and had already chose that very same account as before.  If I replied "no" I would get dropped at the main menu, having to press "Login" again, find the service, and scroll to the login I wished to use.  

However, I have another website for which I have _two_ logins.  When I visit this site - regardless of the "Login Confirmation Prompt" setting - I am presented a choice between the _two_ logins and one isn't automatically chosen and entered. 

### Step by step guide to reproduce the problem
Create **five** accounts for the same website.  Store all of them in the Mooltipass.  Change the "Login Confirmation Prompt" setting to "Off".  Now visit that same site, and observe if one account is chosen automatically (_randomly? last used?_), with the credentials sent, and password sent without any input from you.

Change the "Login Confirmation Prompt" setting to "On" and visit that same site again.  Notice if the same account is now "favored" and you are being asked if you wish to proceed with this selection.  Select "No" and observe being dropped back at the top menu in the device.

Now create _two_ accounts for another website.  Leave  the "Login Confirmation Prompt" setting to "On" and visit the site.  Observe that you are presented a list and asked to choose which of the _two_ you wish to proceed with.  Now change the "Login Confirmation Prompt" setting to "Off" and visit the website again.  Observe that you are again presented a list and asked to choose which to proceed with.  

The "Login Confirmation Prompt" setting should **only** effect Services or Sites for which there is **only one** entry.

### Moolticute Version
v.0.53.1

### Operating System
Mention if you are using either:
- Windows 10
  
### Mooltipass Device
The Mooltipass Mini BLE

limpkin, 2021-12-04:
@CubeCentral I can't reproduce your issue: the device always presents me with a list of credentials to pick from. maybe the credential cache kicked in?

CubeCentral, 2021-12-04:
If you mean the browser itself performed the login before the device was able?  I will double check but I am fairly certain that I’ve disabled that on the browser I most commonly use to access Gmail.  Let me dig into it a bit more

 

               -C

CubeCentral, 2021-12-04:
I tested it just again now, but it is slightly different than I originally stated:

- Woke up Mooltipass and signed in
- Opened up my usual Incognito Mode browser that I always use for gmail
- typed in gmail.com, hit enter...
- as soon as the page was loaded, Mooltipass logged that same credential and password as before.  I didn't even see a list nor was given a chance to choose.

note:  I have seen this work just fine in the past.  Meaning that I would go to gmail.com and it would show me the list of all credentials.

Tested again with a different website, and for one which I have two usernames.  I was shown both of them and prompted to choose.

Not quite sure how to get into gmail with one of the other four accounts...

limpkin, 2021-12-09:
I'm puzzled... can you check that everything is OK with gmail when going in management mode?

CubeCentral, 2021-12-10:
I've figured it out!  Yay!  The explanation and example as follows:
----
Say I have five logins to **gmail.com**:
- cat@gmail.com
- dog@gmail.com
- horse@gmail.com
- emu@gmail.com
- goat@gmail.com

...and it seemed that Mooltipass was always _sending_ the emu@gmail.com credentials (without login prompting on) or _shown selected_ (with login prompting on) and I was not getting a list to choose from.  I was not able to duplicate this behavior on a different domain that also has five accounts.

I had a _eureka_ moment after looking at the list of credentials again and seeing that there was also an entry for emu@gmail.com under **google.com**.  I looked at the address bar while logged into a Gmail mailbox and saw - of course - **google.com** as the TLD and not **gmail.com**.

I deleted the entry for emu@gmail.com as well as the **google.com** service, then re-named the **gmail.com** service to **google.com**.  This way all five of the logins were now under **google.com**.

Testing it with _login prompting_ enabled and with  _login prompting_ disabled now presents me the list of five logins to choose from.

I suspect a password change for the emu.com account was the cause of the desynchronization.  The password may have been changed away from the Mooltipass, then an attempt to login via the Mooltipass was performed with the old password.  Then new credentials were stored with the new password, but for the (now) google.com domain.  That's my guess anyways.

I apologize that I didn't see this sooner!  The entry for **google.com** may have been hidden just off the bottom of the Mooltipass display preventing me from noticing it.

Please mark this issue as completed.  Cheers!

 -cc

closed 2021-12-10 by CubeCentral

limpkin, 2021-12-10:
ohhhhhhh that explains it then :)

No worries :)