mooltipass/minible #314: Mooltipass Mini BLE randomly generating "spacebar" keypress every few minutes.

Technokiddy, 2022-01-14:
### Expected behavior
Mooltipass BLE connected through USB to PC should be idle and not perform keystrokes. when not using the device.
### Actual behavior
When Mooltipass is connected to my PC (Windows 10 and latest FW/Moolticute) the Mooltipass randomly generates a "spacebar" keystroke.
I first encountered this behaviour when watching youtube video's and the video would randomly be paused. Later on I also noticed my computer would be waking up from sleep every few minutes for no clear reason.

After several of these happenings I figured out that once I disconnected my Mooltipass from USB my PC would no longer randomly pause youtube video's or wake up when in sleep mode. Hence my assumption it generates a random keystroke which seems to be a space since thats the hotkey on youtube to pause video's.

### Step by step guide to reproduce the problem
Connect through usb to PC.
Pincode can be filled in or left logged out on the mooltipass device.
Use pc and watch some long youtube video's. Pauses will randomly occur.
Set PC to sleep while Mooltipass is still connected. PC suddenly wake up.

Tested this issue on my desktop (Windows 10) and my laptop )Lenovo Thinkpad T14 (Windows 11). Both systems show the same issue.

### Moolticute Version 
Moolticute: V0.53.7

### Firmware Version
Aux MCU version: 0.69
Main MCU version: 0.79
Bundle Version: 7
Issue also ocurred in Bundle Version: 4 

### Operating System
- Windows 10 / Windows 11

### Mooltipass Extension
N/A - Since it also brings PC from sleep, The Chrome extension can not be causing this trigger.

limpkin, 2022-01-14:
Uh... that's a first. Could you confirm this issue on another computer?
There's no line of code that does periodic key sending... so I'm extremely perplexed

Bloodhoundje, 2022-01-14:
I have confirmed the behaviour on my windows 11 laptop aswell, same for when connecting the BLE to my Android (Samsung Galaxy S10 mini) phone. The moment the BLE is connected through bluetooth on my phone, Apps like for example 9gag suddenly scroll back to the top every few minutes. The moment I disconnect my BLE from bluetooth (or on PC from the USB) the problems instantly stop occuring. And when reconnecting it happens every few minutes again so it is 100% being sent from the  Mooltipass.

I was hoping the latest FW upgrade would have resolved the behaviour but this did not fix it.

limpkin, 2022-01-14:
so this issue is happening through USB **and** Bluetooth?

Bloodhoundje, 2022-01-14:
Correct, I just realized that when I was typing my previous comment and realized I also experienced the issue on my phone. I will be keeping an keystroke event logger running on my pc when idle the next few days to figure out which keystroke it generates and update the issue here.

limpkin, 2022-01-14:
oh yeah that's a great idea actually

Bloodhoundje, 2022-01-31:
I have been running the logger for a while, so far as usual with bugs, it wont happen while I am trying to figure it out :-). Hopefully it will reoccur soon. It might also have something to do with changing USB states that wake the system. 

Will update at the end of this week again.

Bloodhoundje, 2022-02-23:
Finally managed to get the problem to appear again, I dont get a keypress event registered, but the system keeps waking up when the mini BLE is attached. It seems like there is another USB event being triggered by it which is causing the issue. The moment I unplug the Mooltipass the erratic behaviour stops. 

Seems like I have to find a way to log USB events to figure out what is triggering this.

limpkin, 2022-03-28:
hello there! any news? ::)

Bloodhoundje, 2022-03-30:
> hello there! any news? ::)

I have been logging usb events to no avail to what the cause has been so. I will keep monitoring it and hopefully at some point figure out what usb message/package is causing the issue. 

I will close this issue here for now and open a new issue once I have pinpointed the exact cause. (dont want to keep your backlog clogged with this 👍 ).

closed 2022-03-30 by Bloodhoundje

limpkin, 2022-03-30:
no worries, keep us updated :)