mooltipass/minible #316: Enable by default "disable BLE" options

GuillaumeHemmen, 2022-01-31:
### Missing feature

Having the following feature toggled on by default would be nice, so the BLE is "saving" battery by cutting down BT when not needed:

- Disable BLE on Card Removed
- Disable BLE on Device Lock

### Justification
Disable BT radio by default so "newbies" have an already optimised BLE without having to fiddle with the moolticute. This might be more useful for simple users than power users.

### Chalenge
How about existing users who might not want this override by an FW upgrade? 🤔

limpkin, 2022-02-01:
another suggestion from a tester: switch off on bluetooth disconnect.

limpkin, 2022-03-22:
I'm just thinking that if we were to do this then users won't get the computer notification that their device is locked though...