mooltipass/minible #317: minible types wrong characters (again)

schlomie, 2022-01-31:
I recently upgraded from bundle 4 to bundle 7.  Since the upgrade, the minible is exhibiting the same behavior as issue #270.

Passwords containing a `\` are typed out as a `<`.   Passwords containing a `|` are typed out as a `>`

These passwords are displayed correctly on the minible, and interestingly, when autofilled through the mooltipass extension seem to work correctly.  Only when typed do I see the issue.

### Firmware Version
AUX MCU version:  0.69
Main MCU version:  0.79

### Operating System
Void Linux 5.15.17

### Language
Device and User Language: English US
USB Keyboard Layout: USA
Bluetooth Keyboard Layout: USA

limpkin, 2022-01-31:
wow that's surprising... can you double check everything and make sure it isn't a CAPS LOCK related issue?

schlomie, 2022-01-31:
Yup - nothing has changed.  Locale, keyboard layout, everything is the same, except for the new bundle 7.  Also related to #270 and #263, if I switch over to the MacOS (US) layout, it works correctly again.

limpkin, 2022-01-31:
> Sorry, no. I haven't touched the layout on Void. Changing it on the
> minible to use Mac OS (US) instead of USA, then it works.
>
> Referencing
> #270 (comment)
> #263 (comment)

Could you try on another os?
I'm just baffled this is only now reported to us given our extensive
testers program, especially for that widely used layout

schlomie, 2022-02-01:
No luck.  Same result with Arch and Ubuntu 20.10.  Unfortunately I don't have a windows machine to test on.  Once I can get back into the office, I can plug it into one of the various workstations there - but that likely won't be for a while.

Anyway.   It DID work correctly on Android, which is consistent with our discovery findings back in July with #270.  This makes me think it's another CLDR issue.

Last time, you corrected the issue with: [2a80c22](https://github.com/mooltipass/minible/commit/2a80c2258a12774adad7026d16cd6c513d68b29f) and [5c9f423](https://github.com/mooltipass/minible/commit/5c9f423e16cbcd8b4a49f18c34f82453598bda10), properly tying the `Shift` modifier with the correct key.  This fix was good through bundle 3 and 4.

[78410d7](https://github.com/mooltipass/minible/commit/78410d7378b844d7f0578d0f790de54a97eafcf0) appears to be the only commit to `cldr_parser.py` since.  Is it possible this change was not included with bundle 4? (I'm trying to trace the correct tag for 4...)  and is it possible this change affected change you made to force selection of iso C12?

At any rate, this behavior was identified before bundle 3, was corrected and was flawless through bundle 4, and is only now presenting itself again once upgraded to bundle 7.

Here is a video demonstrating the issue on the US layout.  I first display the password on the device, showing the `\` and `|` characters.  I then show it typing out as `<` and `>`.   And then to demonstrate what I mean with the MacOS layout, I switch over to the MacOS US layout and repeat, demonstrating that it does get typed out correctly.

https://user-images.githubusercontent.com/1683217/152009466-b39314f5-df32-472a-97e9-854af2d03048.mp4

limpkin, 2022-02-01:
Wow, thanks for looking into that :).
So on MacOS it seems \ is mapped to D13 and | to shift+D13
On windows/Linux \ is mapped to B00 and | shift + B00, with D13 nowhere to be found (only valid in ISO/IEC 9995-2)

Looking at https://en.wikipedia.org/wiki/ISO/IEC_9995 I read:
"In fact, several layouts (e. g. the US layout), to allow a wider return key, have a key at D13 (shown in green) instead of C12 (shown in yellow). Thus, while they cannot be called “harmonized 48 graphic key keyboards” according to the standard, they still comply to the standard itself. ISO/IEC 9995-3:2010, in referring to the basic layout within its specific scope, does take a possible substitution of C12 by D13 into account. "

It does look like we can hard map C12 to D13 in cldr_parser... what do you think?

schlomie, 2022-02-01:
I think that is accurate.  Glancing around at my various laptops and keyboards - ALL of them have the `| \` key in the D13 position, to the right of `} ]` and above `Enter`

Guess I never gave much thought to the placement of these keys in regards to in standards, outside of the standard QWERTY [...]

At any rate, I think a D13 swap is probably a fairly safe bet.  Are the CLDR mappings just out of date with latest ISO standards?  Would it be safer to create a new "Linux" mapping, copied from the Windows mapping, with the holes filled in?

limpkin, 2022-02-01:
that's.... a tough one. I may make an extra one just for safety indeed...

closed 2022-02-05 by limpkin

kbembedded, 2022-02-07:
I ran in to this as well. I cannot be 100% certain it was the upgrade to v7 due to my use timeline, however.

I use a Linux guest VM in a Windows host via VirtualBox. Starting today (the first time I've used the VM in 2 weeks) I noticed a similar issue of the pipe character in passwords not working correctly. Keyboard layout set to USA. They work fine in Windows, but typed in to the guest VM (miniBLE only attached to host, not guest) the pipe character ended up as a `>` as noted above.

Switching to USA (MacOS) keyboard seems to resolve this in both Windows host and Linux guest and may be a viable workaround for most.

limpkin, 2022-05-03:
for everyone finding this isusue: the "Linux" layout should do the trick :)

schlomie, 2022-05-03:
I can confirm - after upgrading to bundle 8 and selecting the USA (Linux) layout, these characters are typed out correctly.

Thanks again @limpkin !

kbembedded, 2022-05-03:
And the USA (Linux) layout works well in Windows too as far as I can tell.

commit 0361cb6f (2022-02-05): fix #317