mooltipass/minible #318: #002 Contact Support

rastagraffix, 2022-02-14:
### Expected behavior

n/a

### Actual behavior

My minible ran it's battery down and sat on the desk for a few days. I plugged it into the charger about an hour ago and it ticked along fine until a few minutes ago when the display lit up with  ```#002 Contact Support``` displayed.

This seems to pop up every five or so seconds, stays on the display for a couple of seconds, reverts to the normal charging display, and then the whole thing repeats.

I did a quick grep through the source code but I don't see where that message gets generated.

Also, I tried emailing this to support@mooltipass.com but you refused the message :-(

```
Feb 14 13:29:44 orthanc smtpd[67799]: efb538435a64f94b mta delivery evpid=59b670e4d7fa1df3 from=<lyndon@orthanc.ca> to=<support@mooltipass.com> rcpt=<-> source="208.79.93.154" relay="178.32.124.207 (mx4.mail.ovh.net)" delay=10s result="PermFail" stat="554 5.7.1 <support@mooltipass.com>: Relay access denied"
```

limpkin, 2022-02-14:
Hello!

Uh that email bounce is very odd... anything special with your mail configuration?  
Is this 002 reproducible? Are you running bundle v7? next time please keep the bug template :)

rastagraffix, 2022-02-14:
Sorry, firmware 0.53.7.  As for reproducible I'd have to let the batteries drain and then plug it in again. It's still doing it, for what that's worth. Right now I'm waiting to see if I can get an actual charge on the battery. After an hour, it's not looking promising. I'll give it another hour, then I'll pull the plug and let it drain again.

As for the MTA it's a bog standard OpenBSD installation that happily sends mail to dozens of other mailing lists and addresses. That ```554 5.7.1``` is a strong indication that ovh.net is actiely blacklisting me for some reason. You might want to check with them as to why, and ask them to remove the block.

limpkin, 2022-02-14:
can you perform a hard reset?
- unplug USB
- press the wheel for 30 seconds
- plug USB
bundle version is the one displayed in moolticute's about tab :). the strange thing is that we're not using OVH....

limpkin, 2022-02-14:
oh that's it then :).  
when you performed the reset, did you get the start animation?

rastagraffix, 2022-02-14:
This just keeps getting weirder ...

After the first restart I got the startup animation and then the mini went to sleep as per usual. But about a minute later the 002 messages started again.

I did another hard reset an everything was the same -- 002 messages started about a minute in.  This time I tried inserting a card. The mini brought up the 'enter pin' display, then promptly locked up hard.

After the third reset I got the normal startup, but the 002 messages have stopped and this time I was able to unlock the card. Also. the BLE link seems to be working.  I'll leave it on the charger for a while longer.  If I can figure out where I put the USB-C-to-C cable I'll get the mini plugged into the MacBook and see what happens when I probe it over the USB link.

Meanwhile, could you point me to where in the firmware source that error message gets generated? I want to dig around and see if I can figure out what was causing it.

limpkin, 2022-02-15:
can you tell me which bundle version you're running please? :)
here you go: https://github.com/mooltipass/minible/blob/master/source_code/main_mcu/src/main.c#L930

rastagraffix, 2022-02-15:
I left the minible on the charger over night. When I woke up this morning it was back to cycling up the 002 message, just like yesterday.

limpkin, 2022-02-15:
could you post here a screen capture of moolticute's about tab, with your device connected and not encountering 002 error?

rastagraffix, 2022-02-16:
[image]

limpkin, 2022-02-16:
can you migrate to bundle v7 using https://www.themooltipass.com/updates ?
bundle v6 fixes 002s :)

rastagraffix, 2022-02-16:
Not at the moment. See https://github.com/mooltipass/moolticute/issues/979.

limpkin, 2022-02-16:
damn... can you borrow someone else's computer? :)

rastagraffix, 2022-02-16:
Not really. But I do have an old mini sitting on the shelf. Let me see if I can get it fired up again …

limpkin, 2022-03-21:
@rastagraffix have you been able to update your unit?

rastagraffix, 2022-03-21:
Mathieu writes:

> @rastagraffix have you been able to update your unit?

Not yet.  We got hit with a DoS at work and that's been sucking
up all my time.  This past weekend I did get back to working
on the Moolticute FreeBSD port.  I'm planning to get heads-
down in that code this coming weekend, and hopefully get the
port done.  When that happens I should be able to push out
bundle 8 to the device.

--lyndon

cornelinux, 2022-04-02:
I have the same issue. 

#002 error recycling.
However, if I want to check for the versions, the moolticute output is empty.
Serial: 70
Device Memory: 8MB
Aux MCU: 0.58
Main MCU: 0.58
New bundle update available [here.](https://www.themooltipass.com/updates/index.php?sn=70&bundlev=0)

However, I tried to follow this instruction https://mooltipass.github.io/minible/updating_your_device but it looks like nothing happens after 10 minutes. The screen keeps switching between "Remove card" and "#002 Contact Support".

limpkin, 2022-04-02:
@cornelinux can you perform a hard reboot to try to have a chance to flash the new bundle?
- unplug from USB
- press the wheel for 30 seconds
- plug back to USB

cornelinux, 2022-04-02:
Hi @limpkin , thanks!
I have pressed the wheel for quite a time. I also think the battery is broken. So should something happen after the 30 secs?

limpkin, 2022-04-02:
the display should go dark and the device power-off. bundle v8 fixes battery issues :)

cornelinux, 2022-04-02:
Well, my devices immediately goes dark when I unplug the cable! 
This is how broken the battery seems to be.

limpkin, 2022-04-02:
I see. When you connect it back, I'm guessing it displays the boot animation?
how long until the 002 error comes back? not long enough to perform the update procedure?

cornelinux, 2022-04-02:
OK, I simply press-hold the wheel while the cable was plugged in. It turned dark. I replugged the cable and now...
...firmware update is running.

Bundle check........................

Looks good so far.

cornelinux, 2022-04-02:
Great - Firware Version 8, now!
It claims battery at 30% (before it claimed fully charged).
Till now, `#002?` did not occur.
Thanks so far!

I was using my mooltipass mini (non-ble) all the time, but I will investigate the ble more, now.

cornelinux, 2022-04-02:
Oups, `#002` is back.

Moolitcute displays the batt like this:
[image]

limpkin, 2022-04-03:
@cornelinux give it a little time to charge fully, that first charge can be quite long :).
After that, let's check for 002s... if they continue, i'll send you a replacement device :)

cornelinux, 2022-04-05:
@limpkin thanks a lot.
The device has charged - however, I am not sure how strong the battery is.
It looks like the firmware update fixed the 002 issue. I somehow hijacked this issue, so I do not know if anyone should close it.

limpkin, 2022-04-05:
Awesome!
i'll close it once @rastagraffix has updated his device and confirmed that it indeed fixed his issue :)

rastagraffix, 2022-04-06:
> i'll close it once @rastagraffix has updated his device and confirmed that it
>  indeed fixed his issue :)

Excellent!  I've been bogged down with medical issues for the last
couple of weeks, so I'm in catch up mode at the moment.  I have the
new moolticute downloaded and installed now but I have to get a
charge on the minible, so I won't be able to test anything until
later tonight.  I'll let you know the results tomorrow.

rastagraffix, 2022-04-07:
The BLE has been charging all day but it's still doing the #002 dance and hasn't yet gotten much of a charge onto the battery -- only maybe 15%.  So I'm going to leave it on the charger overnight to try to get some more juice in there before I attempt to flash the new firmware onto it in the morning. So, tune in tomorrow ... Same Bat Time! ... Same Bat Channel! ...

limpkin, 2022-04-07:
i hope you're doing better :)  
Please note you don't need to have a full battery to update your firmware: simply keep the shift key pressed when you click the upload button :)

rastagraffix, 2022-04-07:
This thing hates me. Now when I connect it via USB to the Mac I get nothing on the BLE display. Moolticute knows there is something plugged in, but it only gets as far as reading the battery charge - it can't see any of the rest of the BLE hardware, so even with a working moolticute build now, I still can't flash the new bundle.

Note that I've tried this with multiple USB cables on multiple ports, so it's definitely the BLE that's at issue.  For a lark I tried plugging in the dev board, but that just displays "No Aux MCU" on the OLED, and moolticute doesn't see anything being plugged in at all.

At this point I'm becoming pretty convinced I just have bad hardware.

[image]

rastagraffix, 2022-04-08:
I'm still working on the FreeBSD 'cute' port, so I'll carry on with that for now. Who knows? I might have better luck talking to FreeBSD than to macOS.

limpkin, 2022-04-08:
Could you get in touch with us at support(at)themooltipass(dot)com? I think
we will send you a replacement device....

closed 2022-04-26 by limpkin