mooltipass/minible #320: #002 Contact Support after Crash (reproducabel)

Jan-NiklasB, 2022-02-26:
### Expected behavior
When pressing the wheel and scrolling back in "Login", I expected to get to the end of the Credential-List

### Actual behavior
As soon as I press the wheel and scroll back the multipass reboots and shows "#002 Contact support"

### Step by step guide to reproduce the problem
(I don't know if it matters, but regarding to #318 my multipass was also sitting in my backpack for a few weeks)
Startup the multipass
Insert the card and PIN
navigate to "Login"
Press the wheel and scroll backwards

Crash happens

### Firmware Version
AUX MCU version:  0.69
Main MCU version:  0.79
Bundle version:  7

### Moolticute Version - If Involved
0.53.7

### Operating System
Mention if you are using either:
Linux Mint 19 Cinnamon

limpkin, 2022-02-27:
Hello there!
This is actually due to a previous generation import into the mini BLE.
Could you:
- enable developer logs in moolticute
- restart moolticute
- enter management mode
- send us the logs at support[at]themooltipass(dot)com?

Thanks!

Jan-NiklasB, 2022-02-27:
@limpkin Mail is send ;)

limpkin, 2022-03-07:
have you received our reply? did it fix the issue? :)

Jan-NiklasB, 2022-03-07:
Hey,

Sorry totaly forgot to answer... I was on holidays with my girlfriend until yesterday.

I will test it asap and give you a feedback.

Thanks for your great support!

Jan-NiklasB, 2022-03-17:
Hey, I finally had time to test your suggestion now.

And well, what should I say, I don't really know why or how, but it seems to work...

Thanks Mathieu!

For anyone else having this problem:

After sending my debug log to Mathieu he gave me a hint to delete, save and re-add the credentials of a specific website.
After doing this the problem now seems to be gone.

Maybe @limpkin can clarify what to look for in the logs.

limpkin, 2022-03-17:
so this issue is caused by importing old generation databases to the mini BLE. In some cases (we're not sure why) a given service is present twice in the database. When this is detected on the mini BLE (scrolling up), the device reboots itself.

closed 2022-03-17 by limpkin

Jan-NiklasB, 2022-03-17:
Fun fact, I only im and exported databases between my devices which always had the same FW version until never loaded a older backup in a device with newer FW...

limpkin, 2022-03-17:
hmm, but you did have a mooltipass mini or original, right? :)

Jan-NiklasB, 2022-03-17:
Nope, only mini ble

limpkin, 2022-03-17:
Oh then that's a point of worry for us, as I'm guessing there are some
cases where moolticute might create a service during credential merging.
I'll check on our side.