mooltipass/minible #323: (macOS) Encrypted APFS password is rejected if sent over USB direct to password field

MaplesThroughWindows, 2022-03-07:
### Expected behavior
Choosing to enter a password credential over USB directly into an APFS password request dialogue should be successful, assuming the correct keyboard layout is chosen and the password is stored accurately on the Mini BLE.

### Actual behavior
The APFS password is rejected.  The same credential will be accepted if it is typed from the Mini BLE into a text editor first and then copied-and-pasted into the same APFS password dialogue.

### Step by step guide to reproduce the problem
1) Attempt to mount an Encrypted APFS volume
2) Attach the MiniBLE to the host computer over a USB connection
3) Confirm that the correct Keyboard Layout is chosen
4) At the APFS password entry dialogue, use the MiniBLE to type the appropriate password credential
5) When the password is rejected, use the MiniBLE to type the same password into a (secure) text editor
6) Copy and paste that password exactly as typed into the APFS password dialogue
7) Password will be accepted.

### Firmware Version
AUX MCU version:  0.66
Main MCU version:  0.77
Bundle version:  5

### Moolticute Version - If Involved
N/A

### Operating System
MacOS 10.13.6  

### Mooltipass Extension
N/A

limpkin, 2022-03-09:
Could you try increasing the delay between key presses?

MaplesThroughWindows, 2022-03-09:
I made a mistake in my original bug report.  The operating system is not 
macOS 10.13.6, but macOS 11.6.4.  I believe that's currently the most 
recent version of macOS Big Sur.

Following your request I doubled the delay from 15 ms (the default) to 
30ms, but I'm afraid this doesn't resolve the issue.  The outcome is 
exactly as before.

It's as though the APFS password dialogue is somehow interfering with 
the password entry from the Mooltipass in a way that the text editor isn't.

I tried assuming that, perhaps, the first character from the Mooltipass 
wasn't being sent to (or received by) the APFS password field so I 
entered it manually before sending the password, but this didn't resolve 
the issue.  I also tried assuming that the last character wasn't being 
sent (or received) and added it to the end manually after sending the 
password from the Mooltipass: again, this didn't resolve the issue.

Without some way to see exactly what text the APFS password field is 
receiving, I'm not sure how to troubleshoot this further, and I imagine 
that, for security purposes, grabbing that text will be very difficult 
if not impossible.

Mathieu:

limpkin, 2022-03-09:
could you change the password to only type a-z A-Z 0-9 characters? I'm wondering if it's something related to ' " or any other deadkey funky stuff.
I know for a fact that macOS has a tendency to remap key strokes depending on the application the key strokes are sent to...

rastagraffix, 2022-03-09:
MaplesThroughWindows writes:

> It's as though the APFS password dialogue is somehow interfering with =0D=
>
> the password entry from the Mooltipass in a way that the text editor isn'=
> t.=0D

This could be MacOS' internal "security" getting in the way.  I have
noticed that some (not all) of the system-level password dialogs are
not friendly with any password managers (1password in my case, but
others have reported the same thing with other password managers).
But it's not consistent, and even if it was I can't figure out
how to debug the behaviour -- MacOS goes out of its way to
keep you from peering into its innards at that low a level :-(

--lyndon

MaplesThroughWindows, 2022-03-09:
I did some testing.  I have now isolated the problem to the presence of the ~ symbol (tilde) in the password - specifically as a discrete character, not as a diacritic.

I have also confirmed that this issue exists on macOS 10.13.6 as well as 11.6.4.

My testing process, for the record, was to create new APFS Encrypted volumes with each password beginning with the same string of alphanumeric characters (aAzZ09) in each case.  I then gave each volume's password a unique symbol at the end.  I tried the following characters:

> Ampersand &
> Equals =
> Hyphen -
> Dollar $
> Hash/Pound #

These symbols were all accepted without issue by the APFS password prompt.  Only the ~ (tilde) prompted the error.

limpkin, 2022-03-17:
so basically we're parsing https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr-keyboards-37.0/keyboards/osx/en-t-k0-osx.xml and taking the "simplest" combination for a given char to type

In the case of ~ I'm seeing these lines:
- shift+E00: https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr-keyboards-37.0/keyboards/osx/en-t-k0-osx.xml#L60
- opt + b06: https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr-keyboards-37.0/keyboards/osx/en-t-k0-osx.xml#L205
- opt + shift + b06 (tagged as no transform!): https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr-keyboards-37.0/keyboards/osx/en-t-k0-osx.xml#L256
- opt + caps + b06 (tagged as no transform): https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr-keyboards-37.0/keyboards/osx/en-t-k0-osx.xml#L307

Setting that bool to True https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr_parser.py#L791 I'm seeing that the Mini BLE indeed uses shift + E00 for the TILDE. 

If I follow you correctly, that effectively means an error in the CLDR as the shift + E00 combination should be tagged as "no transform".

MaplesThroughWindows, 2022-03-22:
I don't have the technical knowledge to comment on your last post.  If you need any more testing done on this issue let me know.

limpkin, 2022-03-28:
so I'm looking at how to type the tilde symbol for MacOS and finding online posts that do say that a space is required after the combination: https://apple.stackexchange.com/questions/135135/how-to-write-out-a-normal-tilde-sign-in-mac
What **exact** layout are you using @MaplesThroughWindows on MacOS?

MaplesThroughWindows, 2022-10-11:
The laptop I experienced the issue on is a British English macOS 
keyboard.  The output of locale indicates I'm using en_GB.UTF-8 for all 
LC settings.

Mathieu: