mooltipass/minible #324: Confused messaging on purpose of device-specific cryptographic hash

MaplesThroughWindows, 2022-03-10:
In the email received after requesting a cryptographic hash, it states that the purpose of it is to verify that the device is "genuine".  I'm assuming that the main body of the text of this email is a template sent to all Mini BLE users.

In the manual, on page 23, it says:

`[T]he security challenge feature […] lets you make sure your device's firmware hasn't been tampered with.`

If the hash verifies that the device is genuine, this is no guarantee that the device hasn't been tampered with.  If the hash verifies the firmware hasn't been tampered with, this doesn't necessarily mean the device is genuine.

This mixed messaging sows uncertainty about the security of the device and the value of the cryptographic hash.

limpkin, 2022-03-10:
> If the hash verifies that the device is genuine, 
As the manual mentions, it does check that the device **firmware** hasn't been tampered with.

>  If the hash verifies the firmware hasn't been tampered with, this doesn't necessarily mean the device is genuine.
The phrasing indeed doesn't mention anything about physical integrity (embedded sniffer or else). I'd argue that mentioning physical integrity here would sow even more uncertainty.

closed 2022-03-10 by limpkin

MaplesThroughWindows, 2022-03-11:
Whether a device is genuine or not is not the same as its physical 
integrity.

I don't think anyone would reasonably expect a cryptographic hash to 
verify physical integrity.  There are security stickers for that.

I don't think it's reasonable to close this Issue without confirming 
that the template email will be emended to accurately reflect the 
purpose of the cryptographic hash.

Mathieu:

limpkin, 2022-03-11:
Could you suggest an email template that you think is reasonable?

MaplesThroughWindows, 2022-03-11:
Sure.  I can only offer an English template though, I don't speak any 
other languages.

The message I received was as follows:

```
Hello there,

Thanks for backing the Mooltipass project and wanting to make sure that 
your device is genuine using the Security challenge feature.
To do so, please follow these steps:
- if not done already, update your device to bundle v7 using 
https://www.themooltipass.com/updates/
- in moolticute, click on the "About" tab on the top right corner of the 
application
- in the "security challenge" box, in the password field, enter the 
following string:
- click on validate: you should see this string appear:

Please note that the code we provided you is only valid once. Any other 
attempts of re-using this code will lead to an error.
Have a great day!
Mathieu
```

My suggestion is:

```
Hello there,

You can confirm that the firmware on your Mooltipass Mini BLE has not 
been tampered with and is exactly as intended by following the steps below:
1) If not already done, update your device to Bundle v7 by downloading 
the update from https://www.themooltipass.com/updates/ and following the 
instructions there;
2) Open the Moolticute application and click on the "About" tab in the 
top-right corner of the application window;
3) In the "Security Challenge" section, enter the following string into 
the Password field:
4) Click on "Validate", and check that the new string that is displayed 
matches this one:

If the validation strings match: congratulations - you have successfully 
verified the authenticity of your device's firmware!
If the strings do not match, then <I HAVE NO IDEA WHAT THIS ACTION 
SHOULD BE, BUT THIS SHOULD PROBABLY BE SPECIFIED IN THE EMAIL>.

Please note: the code supplied in step 3 is only valid to be used once. 
Any attempt to re-use this code will cause an error message to be 
displayed instead.

Have a great day!
Mathieu
```

Just some comments on the changes I made:
1) I took out the reference to "backing the Mooltipass project" because 
not all purchasers are backers - I didn't back the project, for example, 
I'm just a customer.
2) The email really should specify what to do in the event the Security 
Challenge fails, for whatever reason.
3) I think the email should say more about what the error message will 
be if the code is entered more than once, and what to do if, for 
whatever reason, a user needs to re-perform the Security Challenge process.

Is that useful?

Mathieu:

limpkin, 2022-03-11:
very much, thanks a lot :)