mooltipass/minible #325: (Security) Supplying device serial number when requesting an update opens users to targeted malicious firmware

MaplesThroughWindows, 2022-03-10:
Is supplying the device's unique serial number an unavoidable step for downloading a firmware update?

With serial numbers linked to personally identifiable information, this requirement makes it trivial for a suitably equipped malicious actor to target malicious firmware at a specific user's device(s).

The cryptographic hashes (whose purpose is currently unclear - see Issue #324 ) are sent over unsecured email and so could easily be intercepted and re-written by the same malicious actor.  Or Mooltipass could be forced to supply modified firmware by a state actor, etc.

Basically, if it's not strictly necessary, I think this requirement should be removed.

If it is necessary, would it be possible to explain why?

limpkin, 2022-03-11:
Hello,
- cryptographic hashes are sent over the channels that customers prefer. For example, I've personally sent them over the phone, in person, or over messaging apps of the customer's choosing
- I hope we've clarified the purpose of these hashes by replying to issue #324

> With serial numbers linked to personally identifiable information, this requirement makes it trivial for a suitably equipped malicious actor to target malicious firmware at a specific user's device(s).

I don't understand your sentence. To download your firmware update you need to enter your email, your bundle version and your serial number. If any attacker has access to these it can indeed download that firmware update... but I don't see what that would be an issue as this firmware update is cryptographically signed.

MaplesThroughWindows, 2022-03-11:
The confusion appears to be do to less with implementation and more with 
documentation.

The manual specifically states that users should send an **email** (page 
30 Section I).  If you're happy to use any reasonable communication 
method a customer or backer feels meets their needs, then that should be 
in the manual, along with advice on how to start that process.  That's 
actually a really positive thing and your flexibility is something you 
should make people aware of.

 >To download your firmware update you need to enter your email, your 
 >bundle version and your serial number. If any attacker has access to 
 >these it can indeed download that firmware update... but I don't see 
 >what that would be an issue as this firmware update is 
 >cryptographically signed.

I suppose the concern is that I've never had to supply information like 
this to get access to a firmware update before, and it seems strange and 
raises what I believe are legitimate questions.

It's all about trusting the **source** of the firmware.  At the moment, 
anyone who downloads an update from the Mooltipass project has to reveal 
precisely who they are and which device the firmware is for.  If the 
Mooltipass project gets compromised for whatever reason (and that would 
automatically include the firmware signing key), then the firmware that 
is downloaded can be pre-prepared for a specific person, i.e. targeted. 
  That's assuming you're not already a CIA front like Crypto AG (kidding!).

I'm not sure I see the purpose of having to supply this information just 
for an update in any case, which is why I asked if you could explain. 
Is it because you have to tailor the firmware for specific batches of 
devices?  If so, it would be better to ask users to supply an identifier 
that is less specific but still allows offering the appropriate 
firmware.  Regardless, the email address seems superfluous - aren't all 
Serial Numbers linked to their purchasers on your internal database?

Mathieu:

closed 2022-03-17 by limpkin

MaplesThroughWindows, 2022-03-22:
Hi,

I fixed the link - just a missing trailing / .

Did you mean to close this issue though?  The main point, as described in the subject line, hasn't been addressed.

MaplesThroughWindows, 2022-04-01:
> Closed #325 <https://github.com/mooltipass/minible/issues/325> via
> cd48394

I'm afraid I won't be engaging with this project further until this 
issue receives a proper response.  You closed it with a commit that had 
little to do with the subject and it remains unanswered.  Your apparent 
lack of willingness to clarify the matter reflects poorly on the whole 
project, over and above the nature of the issue itself.

I don't think I'm being unreasonable.  If someone calls me out of the 
blue and asks for my credit card information, alarm bells start ringing, 
and I become suspicious, because handing over these details opens me to 
malicious action without good reason.  The request for serial number and 
email address to download a presumably generic firmware update prompts 
the same response.

Mathieu:

limpkin, 2022-04-01:
Hello,

We're very sorry to hear that.
Following your feedback we have updated our user manual and emails we send to our customers, and we were therefore hoping that it would be perceived as a sign of good faith and openness for feedback.
You did mention:
> If the Mooltipass project gets compromised for whatever reason (and that would automatically include the firmware signing key), then the firmware that is downloaded can be pre-prepared for a specific person, i.e. targeted.
I'd say it's extremely easy to purchase a Mini BLE device while staying anonymous. You could use an anonymous email service, have your unit delivered to a PO box, so that hypothetical state actor would never know who this device is for. When you'd want to download your firmware update, you'd simply provide the email address that isn't valid anymore.

I personally don't think it is fair comparing a request for an email address (that can be anonymous) with a request for credit card information.
If this particular issue is the only point that is preventing you from engaging with the Mooltipass project anymore (and not our source code, security implementation, documentation, ecosystem, GUI interface, features), I hope it is not too pretentious from me to say that the job we have done (so far) is OK.

MaplesThroughWindows, 2022-04-19:
Let's just start at the beginning here.

1) Why do you require users to submit an email address when requesting their firmware update?

2) Why do you require users to submit the serial number of their device when requesting their firmware update?

By the way, you should really re-open this issue, as none of the main points have been responded to, and it would give it more visibility so that other users can comment on this.

limpkin, 2022-04-19:
Hello,
Among others, we'd like to save our server bandwidth and therefore need information that only the device user knows.
I hope this answer is satisfactory.

MaplesThroughWindows, 2022-05-16:
So the reason you're choosing to share is that it's a traffic-limiting 
measure?

Would you be willing to share the other reasons?  If not, are you 
willing to offer a broad reason why?

You have to be aware that being so evasive on this topic is harming the 
way your project is perceived, especially when you've attempted to be so 
open and transparent about other aspects.

Mathieu:

limpkin, 2022-08-05:
The reason I'm choosing to share is that it indeed is a traffic-limiting measure.
Another one is that, even though our update signing mechanism has been checked by qualified individuals, we simply don't want to take the risk to allow a given individual to download **all** update files for our Mini BLE fleet. 
Thanks a lot for bringing this point to our attention.

My1, 2023-02-20:
considering there apparently have been issues with the e-mail to SN association tho, I would propose that the MP brings up some kinda value like for example a shorter version of the verify hash you can enable, that way one would not have to enter an email address neither could all the updates be downloaded.

commit cd483948 (2022-03-17): user manual update for android app, fix #307, fix #325