mooltipass/minible #326: Ios/Mac funny bluetooth behaviour

KDederichs, 2022-03-14:
### Expected behavior

The device remembers the bluetooth connection and seamlessly reconnects

### Actual behavior

While iOS/Mac knows the MP is paired it shows as disconnected/connected but not really doing anything some times.
It seems to happen after you don't use it via bluetooth for a while?

Only way I found to get it back to work was to clear bluetooth pairings on the MP and the Mac/iOS and then reconnect it which is a bit tedious imo.

### Step by step guide to reproduce the problem

### Firmware Version
AUX MCU version:  0.69
Main MCU version:  0.79
Bundle version:  7

### Operating System
Mention if you are using either:
- MacOS  
- iOS

limpkin, 2022-03-14:
can you update to bundle v8, which should actually fix this issue?
https://beta-updates.themooltipass.com/

KDederichs, 2022-03-14:
I flashed the update and I'll be testing if it still happens, I'll let you know in a day or two :)

KDederichs, 2022-03-15:
it doesn't seem to disconnect anymore but battery drain is rather fast I think?

KDederichs, 2022-03-15:
[image]
maybe that's why, seems like it's giving me those every minute or so?

limpkin, 2022-03-16:
wow.... that's.... something.
that's a good pointer to continue my investigation :)

KDederichs, 2022-04-01:
@limpkin 
On the original issue, it's still not 100% solved.
I just tried to connect my iPhone to the mooltipass (after it discharged) and the mooltipass just showed "Too many failed connections" and I have to delete the pairing again for it to connect.

KDederichs, 2022-04-01:
Actually it only worked after forgetting the pairing on the iPhone as well and turning the Bluetooth off and on on the Mooltipass again

limpkin, 2022-04-01:
is this with the new bundle? the changes we made in bundle v8 may indeed force you to redo all of your pairings on macOS/iOS devices

KDederichs, 2022-04-01:
Yeah with v8.
But the original pairing was with v8 as well I think (tbh not 100% sure if I repaired it on that device after I flashed it)

KDederichs, 2022-04-11:
Yup 100% with v8 as well @limpkin just had that problem again

limpkin, 2022-04-11:
wasn't able to connect again? which bluetooth transceiver are you using?

KDederichs, 2022-04-11:
yeah it kinda forgot it was connected again (had to reset connected devices, forget it on phone and all the stuff described above).

And don't know which one it is, what ever they stuffed into the iPhone XS Max

limpkin, 2022-04-26:
not forgetting you @KDederichs ! please keep updating this ticket whenever you have the issue again.

KDederichs, 2022-07-08:
Happend again, also made a small Video how the iPhone behaves during it:

https://user-images.githubusercontent.com/24696606/178046025-ae1363d8-ca21-4a5c-92a6-a45cdc1ea6f1.mp4

The first press to reconnect is just trying it normally
Then after forgetting the device I deleted pairing on the Mooltipass
And the 2nd paring (the successful one) I disabled and enabled Bluetooth on the Mini BLE

limpkin, 2022-07-08:
could you update to bundle v10 on beta-updates.themooltipass.com ? :)

KDederichs, 2022-07-09:
Haha I'd love to but I think you guys changed the keybind for the update tab on Mac and it's not activating (Shift +F3)

limpkin, 2022-07-09:
it should still be only shift-f3 on the latest moolticute beta :)

KDederichs, 2022-07-09:
Ok, found it.
Say is there a way to skip the tutuoral when flashing the device? That's been something that's been bothering me for a while (might be worth an extra issue)

limpkin, 2022-07-09:
not at the moment i'm afraid.

KDederichs, 2022-07-10:
With bundle version 10 bluetooth seems to randomly turn itself off on the BLE (it connects so far after I turn it on though), which probably should not happen

limpkin, 2022-07-10:
It does that if a device tries to connect with the mini BLE too many times :) (the "too many connections" message)

KDederichs, 2022-07-10:
Well considering I had it just connected to the iPhone without doing anything that's still a bug.
For some reason it seems to get overwhelmed by just being connected then.

limpkin, 2022-11-05:
@KDederichs any update on that bug? :)

KDederichs, 2022-11-09:
It might be fixed?
But tbh I mostly stopped using the mooltipass (sadly) because there's a lot of friction when you regularly try to use it with different devices (in my case an iMac, macBook, iPhone and iPad)...

limpkin, 2022-11-09:
oh really... could you expand on what the main friction was? maybe we can find a trick :)

KDederichs, 2022-11-09:
Sure it's mostly the switching between devices that's annoying.
I don't know if it's just the apps/services I use but I usually have to log into stuff A LOT on different devices, so every time I have to go to re-pair the Mooltipass on most things (cause I usually don't carry the cable around with me since that's the point of the BLE).

That usually involves going into device settings and re-pairing devices (https://github.com/mooltipass/minible/issues/336).

It's just building friction if you do that a few times a day.

If you compare it to software solutions like Bitwarden for example you have nothing of that friction. You just start your auth flow, do the biometrics/password and done.

(Tbh that's also the reason why I kinda let the iOS version of the companion sit around)

limpkin, 2022-11-09:
right, thanks for your explanation!
if you had an iOS companion and leave the bluetooth connection on, would using the scroll wheel shortcut to switch devices be alright though?

KDederichs, 2022-11-10:
It would be an improvement if you don't have to go through (sub) menus to get there, sure.

But imo long term goal should be the handoff between devices. Since if they're already paired they should be considered safe to make requests by default anyways, so it would just make sense for any paired device to make requests to the BLE at any time without manual intervention.