mooltipass/minible #332: TOTP Password

d3t4ash, 2022-04-03:
Is it possible to add new main menu for TOTP Password list in the minible?

It is very handy if we can access the TOTP in the main menu of minible. Please also consider to make this menu in the Moolticute for easy setting (using QR Code scan or type the secret code directly) and displaying this TOTP in the apps.

awilkins, 2023-01-18:
Counterpoint ; if you're using your BLE as part of an MFA solution, putting your TOTP key in the same device as your password turns it into OFA.

Sazoji, 2023-11-14:
> Counterpoint ; if you're using your BLE as part of an MFA solution, putting your TOTP key in the same device as your password turns it into OFA.

The device's master pass is already MFA; Key, pin, and the correct database (not just any device unless loading a backup) are all required to be together at that moment in time to unlock. Having an additional hw key or a phone for auth is not adding an appreciable increase in security, especially if they increase the attack surfaces like a phone can.