mooltipass/minible #334: [Companion] Ability to list credentials for domain

KDederichs, 2022-04-10:
### Missing feature
A command that just lists all available usernames for a given domain (without password). Optimally without user interaction (configurable?)

### Justification
Playing around with trying to build the iOS companion I'm noticing that it would be useful to have a way to just list available usernames for a given domain to present the user with for example the QuickType select.

Having that ability would make for a more user friendly flow with the companion apps (not sure how the android one handles this?)

Flow wise I'm imagining it like this:

Companion app sends list request -> Mooltipass returns none/list of user names -> user selects desired username -> normal get credentials command.

### Workarounds
Using the normal get credentials command, you it's less convenient since the user has to pick up the device to select the desired credential.

KDederichs, 2022-04-10:
oh btw @limpkin let me know if you have a general time frame for this, since the apple auth framework kinda relies on getting that list as far as I can tell :)

limpkin, 2022-04-10:
oh really? One may argue that offering such a feature can become a privacy issue...

KDederichs, 2022-04-10:
yeah I get why it could be viewed as such tbh...
It'd be mainly used for that kind of quick ui I guess:
[image]

I probably could make to without just doing some placeholder messages but that's just inconvenient imo.

Maybe it could be a feature that can be explicitly enabled for people that want it (with a flag that shows if it's on or not)

(screenshot from https://nilotic.github.io/2018/09/07/Implementing-AutoFill-Credential-Provider-Extensions.html)

closed 2022-07-08 by KDederichs