mooltipass/minible #335: Switch-off device after BLE Disconnection

kevin-sinclair, 2022-04-13:
### Missing feature
(general settings in Moolticute) (and relative behaviour on MiniBLE)
[v] Switch Off device after **BLE** Disconnection 

### Justification
Battery life would be increased a lot if we could have the device switching itself off when losing BLE signal (closing/locking laptop for example)
Currently this is only implemented for USB with a checkbox in Moolticute's "general settings":
     [v] Switch Off device after **USB** Disconnection 

**Note**: MiniBLE boots so fast (when you remove the splash screen ;) that it makes no difference between starting it and unlocking it.

**Alternative**: add option to convert 'lock' behaviours to 'off' behaviours across all settings?

### Workarounds
Switching device off manually (when I remember :)

limpkin, 2022-04-13:
implemented in https://github.com/mooltipass/minible/commit/bba22fae4435f53a26675b501e3104d7e93cc01d ;)

kevin-sinclair, 2022-04-14:
Excellent! Thank you.
I'm on Bundle 8 Aux 0.70 Main 0.80 - bba22fa is marked for release testing v0.50/v0.51-testing... but I'm already on Moolticute v.0.55. I can't see the option - where am I going wrong? :)

limpkin, 2022-04-14:
it'll be enabled with bundle v9 release :)

closed 2022-04-17 by limpkin