mooltipass/minible #336: Connect to multiple devices at the same time

KDederichs, 2022-04-16:
### Missing feature
The ability to connect to multiple host devices via bluetooth at the same time would be neat.

### Justification
At the moment in order to use the BLE with different devices via Bluetooth you always need to disconnect them/connect them to switch around which is not that user friendly.
It would be good if you could connect it to multiple devices at the same time and have some kind of auto hand of depending on which device talks to it (Think AirPods and stuff, they do a good job with that).
That way carrying the BLE around to use with multiple devices would be A LOT more convenient.

### Workarounds
Manage the process manually

rastagraffix, 2022-04-17:
> ### Missing feature
> The ability to connect to multiple host devices via bluetooth at the same tim
> e would be neat.

This is something I've been thinking about for a while now, and I
hope to be able to start prototyping something within the next few
weeks.  What I'm thinking of for the first cut is the ability to
set up pairings with multiple devices, and then use the thumbwheel
to select which of the paired devices is the "active" one.

Having the BLE juggle requests from multiple devices authmatically
gets a lot tricker, and there are security implications I need to
think about before starting down that path.  But the first cut with
manual selection should be pretty straight forward to implement,
once I finish setting up my dev environment and resolve some hardware
issues I've been tripping over.

I'm very much interested in any ideas about how the UI for this
should work.  I have my own specific use cases in mind, but I'm
sure there are many others.

--lyndon

KDederichs, 2022-04-17:
>
>> #### Missing feature The ability to connect to multiple host devices via bluetooth at the same tim e would be neat.
> This is something I've been thinking about for a while now, and I hope to be able to start prototyping something within the next few weeks.
> --lyndon

Neat :)

I do disagree that it should be a separate menu setting though since that again breaks the workflow if you have to specifically go to a setting an select the device every time. 
Device switching should be as seamless and non involved as possible imo.

There's two solutions for this I can come up with right now:

1) If the MP detects a connection from a paired but not active device a message could come up asking if Device X should be activated and then proceed with the request depending on the choice.

2) You integrate this into the "main request". If you have prompt mode on you could modify the prompt to also display the device that is asking to perform the action. That way you kind of have the seamless hand off but at the same time the security should not be compromised since the user sees the device that's asking.

Not really sure if the Host Bluetooth name can be trusted though so maybe that would require to manually assign labels to Bluetooth connections (which you have to confirm on device to prevent tempering)

limpkin, 2022-04-17:
Hello All!

To be honest we never really thought this use case would be common and therefore didn't spend much time thinking about it.
Updating the GUI, workflow, Bluetooth low level code won't be a small endeavor @rastagraffix (I don't want to discourage you :) ).

KDederichs, 2022-04-18:
@limpkin Really?
I'd think in the current world where almost everything has integrated bluetooth it would be common use case tbh, since it's just more convenient than always having to have a cable available and plugging it in :)
But yeah I get that changing that probably would be a huge pain.

RChadwick7, 2022-04-18:
I'd like to add my vote for making this a higher priority. I use about 5 laptops and 2 + Android devices regularly. I have two BLE's, and I usually can't use them because they happened to have connected to another device, which one I have no idea. I can try disabling and then re-enabling Bluetooth, and then try to connect, but usually the first connection reconnects first. Sometimes I wind up unpairing, and then re-pairing, but it's obviously cumbersome. While it would also be great if the BLE could tell you which device it's connected to, it would be better if it could connect to 10 devices. Connecting to only one device offers few advantages over simply connecting by cable.

limpkin, 2022-04-18:
we're counting on @rastagraffix then :). All: please thumbs up the issue to manifest your interest for it :)

RChadwick7, 2022-04-18:
I'm not familiar with multiple device Bluetooth programming, so I'll defer to you, or rastagraffix. :) A possible workaround that might be easier to implement would be to connect to one device at a time, but have a list of previously connected devices, display current and other connected devices, and allow to connect to another device. As an example, Say I pair the BLE with a phone and a laptop. The phone is connected to the BLE. I want to use my laptop, and my phone is buried in my work bag. From the BLE, I can tell it to connect to the laptop. Maybe what that really means is that it disconnects from the phone, temporarily blocks the phone from reconnecting, and allows the laptop to connect. Maybe better, allow Moolticute to connect when needed. Say the phone is connected, and I use my laptop. In Moolticute, there's a button that says "Switch Mooltipass to this device?". I'm hoping something could be worked out. Otherwise, from a practical point of view each BLE is useable with only one device, unless if we plug in. If we plug in, we might as well use the non-BLE Mooltipass.

hcjehg, 2022-09-28:
OK, so when you use USB you select the device you want to communicate with, "by plugging into that device". It is not possible to plug into several USB devices... With BLE this becomes more complicated because it is possible to connect to more than one device at any one time. It is already possible however. You can be connected to a USB device and a BLE device at the same time. When you do an operation the interface is specified. I propose to continue with this as basis...

There are (at least) 2 scenarios for this:

Authentication initiated by device connected to the Mooltipass BLE device (MpBLE), either via USB or BLE:
The Identity (ID) of the device that initiated the Authentication must be part of the approval workflow (When user accepts to authenticate on the MpBLE). If connected to BLE and USB, the user would need to see, as part of the approval, the origin of the Authentication. Otherwise the user may inadvertently authenticae on the BLE device when the user wanted to Authenticate on the USB, or vice versa.
The IDs can be either USB or an ID bound to the authentication keys (the LTK i think) for the paired BLE device, so a Friendly name for the paired device.

Authentication initiated by MpBLE:
E.g. the Keyboard Typing Scenario. The first screen after selecting the Login (The Key) on the menu,selects the Credentials. After selecting the Credential a choice between USB/BLE is posed. That menu could be changed to USB/Laptop1/Supercomputer/Phone1/TV1/TV2 (The friendly names for the paired BLE devices) and you would need to select one to send the credentials to. The ID/Friendly name would be the same as the above. The BLE devices currently connected are shown first, then those paired but not connected. This is so you may initiate a connection with a paired device, if, for some reason, they do not connect automatically...

Bluetooth "Disconnect" and "Delete Pairing" would also need to allow selection between Connected/Paired devices.

Did that make any sense? :-)

I'm surely forgetting some details of the workflow... :-)

RChadwick7, 2022-11-28:
As an example, the new Flipper Zero seems to handle Bluetooth well. The device could be sitting in a bag for a week, and I could, using any random paired device nearby, flash the firmware without even knowing where it is. Being on my phone, or any other random device, and using (Perhaps the one with the best signal strength?) any nearby Mini BLE would be helpful. Perhaps it would be better to send the request to ALL nearby devices? Say I have 2 devices. One is in a bag, one is sitting on a table. When Moolticute asks a BLE for a password, it sends it to all devices. Whichever device I find first, I enter the PIN into, and the webform gets it's password.

alanrick, 2022-11-28:
I wonder if this would make the bluetooth handshaking more idiot-proof.  I've often struggled for ages to bluetooth connect with my PC, only to find it connects, when I _disconnect_ on my BLE. Weird. 😨
But understandable if it's disconnecting from another device (e.g. phone) and this then frees the BLE to connect to the PC. And this _disconnect_ sometimes has to be done several times (perhaps it's connected to the iPad without me knowing).

EugeneRymarev, 2022-12-13:
> I'm very much interested in any ideas about how the UI for this should work. I have my own specific use cases in mind, but I'm sure there are many others.

I suggest double clicking on the wheel to change the active device.