mooltipass/minible #337: Last character of 64-character username not output/displayed

hijackeel, 2022-04-18:
### Expected behavior

The complete 64-character username should be output/displayed by the Mooltipass.

### Actual behavior

Only the first 63 characters of the username are output/displayed.

### Step by step guide to reproduce the problem

1) Create credentials in Moolticute with 64-character username, and save to Mooltipass.
2) Select credential on Mooltipass and either output or display credential, and verify that last character of username is missing.
3) Enter Credentials Management Mode in Moolticute, and verify that all 64 characters in username are saved on device.

### Firmware Version
AUX MCU version:  0.69
Main MCU version:  0.79
Bundle version:  7

### Moolticute Version - If Involved
v0.53.7

### Operating System
Windows 10

limpkin, 2022-04-18:
Hello there!  
Oh that's an interesting one.... we support 64 characters long **passwords** but only 63 characters logins :).
Out of curiosity, what made you encounter this use case? do you have such long logins?

hijackeel, 2022-04-19:
I didn't realize usernames were limited to 63 characters.
I just randomly generated one that long because a site allowed it.
Maybe Moolticute should disallow storage of longer usernames, and this ticket should be moved from minible to moolticute?

limpkin, 2022-04-19:
Agreed.