mooltipass/minible #338: Option to not power on while charging - Solar

RChadwick7, 2022-04-18:
This may or may not be practical based on the hardware. My idea was to attach a small 5-6V solar panel to the USB port to trickle charge the battery. Problem is, it appears that, when charging, the BLE wakes up. With a few mA trickle charge, this obviously does more harm than good.

limpkin, 2022-04-18:
I'm afraid this isn't possible as our microcontroller handles the charge algorithm :(

closed 2022-04-18 by limpkin

rastagraffix, 2022-04-18:
RChadwick7 writes:

> This may or may not be practical based on the hardware. My idea was to attach
>  a small 5-6V solar panel to the USB port to trickle charge the battery. Prob
> lem is, it appears that, when charging, the BLE wakes up. With a few mA trick
> le charge, this obviously does more harm than good.

For a future hardware version, a physical power switch might be a
nice thing to have.  There could be times where I would like to
know there is no way anything could wake up the BLE via the radio
interface and start probing ...

--lyndon