mooltipass/minible #340: #002 Contact Support, fixed by Device Settings, Battery Problem?

mario-oberwalder, 2022-05-09:
### Expected behavior
Works as usual

### Actual behavior
Best case, can unlock card, then freeze immediately.
Other behaviour:
- black screen on disconnect
- previous frozen screen on reconnect
- #002 Contact Support
- battery status in moolticute: 0%

Tried letting it charge for about 30 minutes. Same symptoms.

### Step by step guide to reproduce the problem
N/A
Maybe letting the battery run out with BLE enabled.

### Firmware Version
AUX MCU version:  0.70
Main MCU version:  0.80
Bundle version:  8

### Moolticute Version - If Involved
v0.53.7

### Operating System
Mention if you are using either:
- Windows 7
- (Fedora 35)
  
### Fix:
I enabled the following device settings:
- Switch Off Device after USB Disconnection
- Disable BLE on Card Removed
- Disable BLE on Device Lock

After reconnecting/reboot everything seems to work. 
Battery seems to charge.

I'll leave BLE off for now just to be safe.

PS: A backup mooltipass with up-to-date database files is a really really good idea :D

limpkin, 2022-05-09:
hmmmm.... I'm wondering if that's not a device-related issue. can you get in touch with us at support(at)themooltipass(dot)com so we can send you a replacement device to check? :)

limpkin, 2022-06-16:
ping @mario-oberwalder  :)

billgertz, 2022-10-25:
Just received a Mooltipass BLE Mini today and out of the box it wan't charging even after sitting on a 2.5A 5V capable charger and a 5A charger.

Now having the same issue with a ! #002 Contact Support message and the device acting strange trying to recharge. Have sent messages through the Mooltipass.com Contact Us interface. Wondering if I too might have a bad device. Tried battery reconditioning and now the device won't charge past 80%. Since it's not fully charged I cannot update the bundle.

limpkin, 2022-10-25:
hello @billgertz could you update your unit to bundle v8 using www.themooltipass.com/updates ? 
Bundle v8 does fix 002 & battery issues :)

billgertz, 2022-10-26:
Responded at support as well. Pressing shift and select bundle file button at same time changes nothing. Still cannot upgrade the firmware. Um, now what?

limpkin, 2022-10-26:
did you correctly insert the card the wrong way around?

billgertz, 2022-10-26:
Yep

Kind Regards/ Met vriendelijke groet/ Mit freundlichen Grüßen,

Bill Gertz
Senior Consultant
***@***.***

Mobile 	+31 6 422 37005
WW		+1 719 229 4724
Signal	+31 6 422 37005

Blockfish BV | Markt 12A | 4101BX Culemborg | Netherlands
www.blockfish.net

limpkin, 2022-10-26:
damn, it seems you may have found a bug on the bypass mechanism
@deXol do you know what key should be used for MacOS?

limpkin, 2022-10-26:
@billgertz in the meantime would you have access to a non macos computer? AFAIK there's not a way to do this through command line

billgertz, 2022-10-26:
@deXol @limpkin 

I've tried the following key combos already, or at least I think I have ;-/:

 - Option
 - Command
 - Control
 - Control+Command
 - Control+Option
 - Command+Option
 - Shift+Option
 - Shift+Command
 - Shift+Control
 - Shift+Control+Command
 - Shift+Control+Option
 - Shift+Command+Option

billgertz, 2022-10-26:
@delXol @limpkin 
Parallels Windows 10 VM doesn't seem to passthrough whatever USB device. Can you clarify which devices(s)? need to be passed through? Windows 10 system I cannot change tabs, although all other software on the system works without issues. This is approaching silly.

billgertz, 2022-10-26:
@delXol @limpkin
Although I could not change tabs Control+Shift+F3 did bring up the BLE Dev tab. I prepped the device for firmware upload, pressed Shift+Upload Bundle File. And of course it errors with a message:

Error: Upload bundle finished with error

Yes I rebooted the Windows system. Now what?

limpkin, 2022-10-26:
MacOS should still be shift (https://github.com/mooltipass/moolticute/blob/master/src/BleDev.cpp#L174), but our developer let us know in some cases MacOS may remap it to CMD+SHIFT.
@billgertz can you confirm the keys are pressed **while** you click on the upload button?

billgertz, 2022-10-26:
Reattached to Mooltipass BLE Mini closer to MB Pro and now Parallels can see the the device, but MooltiCute cannot see the device. Sigh.

limpkin, 2022-10-26:
We're making progress! I'll update the manual to reflect that potential issue.
Can you confirm the device is showing the "Please remove card" animation?

billgertz, 2022-10-26:
Yes it had and that's why had removed it. BTW - I edited the parent post so you might need to refresh your browser to see that we got the firmware to update.

limpkin, 2022-10-26:
Awesome! Glad you managed to update :).
I indeed recommend on doing a final battery reconditioning... I apologize for all these adventures?

billgertz, 2022-10-26:
Do you need a MacOS QA? Would be happy to help, so no one else goes through this.

limpkin, 2022-10-26:
We do welcome any advanced testing that can be done on MacOS, as it has historically been quite capricious with our software :)

billgertz, 2022-10-26:
Okay - BTW  are using any CI tooling for testing (travis with Linux UI test tooling xvfb, or MacOS UI test tooling atomacosetc)?

billgertz, 2022-10-27:
Still having a wee hiccup - the device didn't charge past 90%, and is still throwing #002 Contact Support. Rebooting does not clear the error. Do I do another battery recondition, or is there something up with the device?

limpkin, 2022-10-27:
I'm starting to think this device might have issues... would you be able to update to the latest beta at beta-updates.themooltipass.com ? let me know if you can't fetch the update as I haven't updated this page for a little while

billgertz, 2022-10-27:
Was able to download when I left the registered email address blank - so the same issue as the production firmware download page. Updating now.

billgertz, 2022-10-27:
Okay, it's been an hour, and still not charging past 90%. So do I try another battery recondition?

limpkin, 2022-10-27:
Hello Bill,

90% max charge can be expected (see user manual).
Out of curiosity, what "time spent" did the battery reconditioning
procedure give you?

Mathieu

billgertz, 2022-10-27:
Well over five hours, but it died with that Contact Support !#002 - let me look at the logs.

limpkin, 2022-10-27:
was bluetooth enabled?

billgertz, 2022-10-27:
Yes

billgertz, 2022-10-27:
> 90% max charge can be expected (see user manual).

Um, not that I can find. Searched manual *charge* is mentioned exactly one:

```
L) Battery Usage Best Practices
The Mooltipass Mini BLE uses a Nickel Metal Hydride Battery (NiMH). For maximum battery longevity, we recommend you refrain from charging your device until its battery is depleted. A single charge typically lasts between 2 and 7 days, depending on how you use your device. When receiving your Mini BLE for the first time, we also recommend performing the battery reconditioning procedure described in Advanced Tab.
```

*90* only once too:
```
- Device size: 90 x 35.5 x 12.5mm
```

Nothing I can find mentions that you can expect a 90% maximum charge.

> can you try disabling it and see if this improves your 002 issues? :)
> […](#)

Since the upgrade to bundle 11 I haven't seen the message yet. Saw what the original poster did for a solution, but I'm waiting to see if it occurs before I go with that.

billgertz, 2022-10-27:
> can you try disabling it and see if this improves your 002 issues? :)
> […](https://github.com/mooltipass/minible/issues/340#)
@limpkin Arggg the !https://github.com/mooltipass/minible/pull/2 Contact Support message just returned while connected to USB. This time the device required a reboot as it was locked up. The battery charge was at a full 90%. It doesn't seem I am having discharge issues. However, did what @mario-oberwalder suggested:

I enabled the following device settings:

Switch Off Device after USB Disconnection
Disable BLE on Card Removed
Disable BLE on Device Lock

limpkin, 2022-11-05:
Hello @billgertz and sorry for the time it took us.
Could you update to bundle v12 using beta-updates.themooltipass.com ? I'm pretty sure I fixed that 002 there :)

limpkin, 2023-01-02:
pinging @billgertz  :)

CGuy-1, 2023-10-13:
I've had that #002 Contact Support error on my first miniBLE device for a while now. It still seems to work with it showing, also get the #011 Contact Support error was well. My first device has an issue with the scroll wheel and the battery can't hold a charge anymore no matter how many time I run the reconditioning on it. I suspect lots of use and periods where it was on when I thought I turned it off draining the battery to 0% all too often is the cause.

I don't think the battery indicator in Moolticute is very precise as it doesn't always react when I change devices or when its been charging for a while and the value doesn't seem to move but the battery icon on the device does.

limpkin, 2023-10-13:
@CGuy-1 could you maybe try bundle v13 at beta-updates.themooltipass.com ?