mooltipass/minible #343: Mini BLE types "|" as ">" on Linux and "±" on Mac

hijackeel, 2022-05-12:
### Expected behavior
The Mooltipass should type stored passwords correctly on any OS.

### Actual behavior
On Linux, pipe comes out as closing angle bracket.
On Mac, pipe comes out as plus-minus sign.

### Step by step guide to reproduce the problem
1. Use Moolticute on Windows 10 to generate and store a password containing a pipe on the Mooltipass.
2. Display password directly on Mooltipass and verify that the pipe is displayed correctly.
3. Use Mooltipass to type password on a Windows computer or Android phone and verify that the pipe is typed correctly.
4. Use Mooltipass to type password on a Linux computer and verify that the pipe comes out as ">".
5. Use Mooltipass to type password on a Mac and verify that the pipe comes out as "±".

### Firmware Version
AUX MCU version:  0.70
Main MCU version:  0.80
Bundle version:  8

### Operating Systems
Gentoo Linux
MacOS

limpkin, 2022-05-12:
did you select the right layouts for macos and Linux in moolticute's settings or in the device itself?

hijackeel, 2022-05-13:
I totally forgot about that! I haven't needed that setting until now. I just tested "USA (Linux)" and it worked.
Sorry about the ticket. You can close it. :)

P.S. I don't want to open too many tiny feature requests, but a small thought: 
The layouts on the Mooltipass are organized first by country, and then by OS, so you have to scroll around to switch OS, and one might not even realize the other OS layout(s) exist. I think it would be better to have it organized purely alphabetically, so for example "USA", "USA (Linux)", and "USA (MacOS)" are right next to each other. I noticed that the list in Moolticute is actually already organized this way.

limpkin, 2022-05-13:
awesome!
that makes sense... but do feel free to add feature requests :)

closed 2022-05-13 by limpkin