mooltipass/minible #345: Mooltipass Enters "Going to Sleep" Loop Immediately After Charging Sometimes

stonemirror, 2022-06-16:
### Expected behavior

Charge the Mooltipass to 100%; take Mooltipass off USB; Mooltipass goes to sleep and STAYS that way, 40% of charge isn't reported to  have instantly evaporated.

### Actual behavior

Occasionally after charging and removing the device from USB at 100% charge, the Mooltipass displays the "Going to sleep" banner, and then immediately wakes up, over and over and over. I power down the device (by pressing the clickwheel until it shuts down). When I restart it, the charge level is at 60%.

### Step by step guide to reproduce the problem

It seems to be random and intermittent.

### Moolticute Version

v0.55.0

### Operating System

- MacOS Catalina (v10.15.7)
  
### Mooltipass Device
- The Mooltipass Mini BLE (s/n 2732, Bundle v8)

limpkin, 2022-06-16:
hmmm that's very odd. I'm guessing bluetooth is enabled in that scenario? is it connected to a device when this happens?

stonemirror, 2022-06-27:
Yes, Bluetooth is enabled, it's connected to my Mac Mini desktop system.

I think I screwed up and created a problem, however: when bringing up Moolticute at one point, I noticed a link to an updated bundle, didn't use it immediately, but then went to the firmware update site, and downloaded a v8-to-v10 from there, using my serial number and credentials.

The update proceeded normally, from all appearances, but now the Mooltipass is stuck in the quick tutorial, just going from the "Welcome to the Mooltipass Mini BLE" screen to the "This will be quick" screen and back to the Welcome screen. I can't get it into an operational mode at all at this point. Any ideas?

My bad here.

stonemirror, 2022-07-01:
Update: since updating the device to the v10 Bundle, I'm not seeing this problem any longer. Typically, I would see it if I did the following, as nearly as I can tell:

— Disconnect the fully-charged device from USB
— Enter my PIN
— Select "Lock" from the main menu

I did NOT see it if I never entered my PIN or entered the PIN but did NOT select "Lock" from the menu. In those cases, after a short interval, the device would display the "Going to sleep" screen, go to sleep, and STAY that way.

Anyway, if I don't see it again in the next several days, I'll close this.

limpkin, 2022-07-01:
awesome :)

stonemirror, 2022-07-02:
Okay, I've just seen the same behavior again:

— Disconnect device from USB
— Enter PIN
— Select "Lock" from main menu

Rather than going to sleep and staying asleep, the Mooltipass displays the "Going to sleep" screen, goes black for an instant, and then wakes up, displaying the main Mooltipass screen (Key, logo, etc.) After a short interval, the "Going to sleep" screen redisplays, and the entire process repeats.

Workaround is to enter the PIN and NOT select "Lock" from the main menu, but to simply allow it to go to sleep from the unlocked state.

limpkin, 2022-07-03:
I can't seem to reproduce this bug... is the device connected through Bluetooth when this bug happens?

limpkin, 2022-11-05:
this bug was fixed here: https://github.com/mooltipass/moolticute/commit/c6b10440e9a204853ff5bf74046ff77a0ca78741

closed 2022-11-05 by limpkin