mooltipass/minible #348: PIN entry through host computer.

arcaartem, 2022-07-24:
### Missing feature
Give users an option to allow PIN entry through the host computer instead of the wheel on the device when it's connected through USB (and BLE if that's even possible). 

That is, when the device asks for the PIN to unlock the database, with this option enabled, the browser and/or the app on the host prompts an input field for the PIN *as well*. 

### Justification
It may be subjective but to me it feels a bit cumbersome and time-consuming to enter an 8 digit PIN number with the wheel on the device every time I need to enter a password on a website. Having an option to provide PIN input through the host computer would be incredibly convenient. 

### Workarounds
N/A

Jan-NiklasB, 2022-11-13:
Would come in handy on the cost of security...
Someone knowing that you own a mooltipass and wanting to get your credentials could try to install a keylogger.

Sure, that would log also all keystrokes done by the mooltipass, but it wouldn't give access to ALL credentials stored on the mooltipass...

arcaartem, 2022-11-13:
Yes you're right: it's a trade-off. Which is why I suggested it as an option. If a user wants to trade off a bit of security for additional convenience, they can choose to enable this option on the device.

Jan-NiklasB, 2022-11-13:
Okay, but must be mooltipass-side only, so it's not possible to enable this feature from the multicute.

arcaartem, 2022-11-13:
Exactly, which is why we're having this discussion on MiniBLE repository and not on the Moolticute one ;)

michaelni, 2023-01-18:
> Yes you're right: it's a trade-off. Which is why I suggested it as an option. If a user wants to trade off a bit of security for additional convenience, they can choose to enable this option on the device.

It would be possible to enter the Pin on the host while maintaining security, in a way similar to how the trezor is doing it. The MiniBLE would display a 4x4 grid with the 0 to F digits randomly assigned. While the host would show a blank 4x4 grid one can click on. The host would know the x/y coordinates of each of teh 4 clicks from the user but not what they mean as their values are different for each pin entry. The host would then send these 4 x/y coordinates to teh MiniBLE which then could interpret it with what it showed the user on its display
That would be a little less convenient than entering the pin directly but should still be a bit quicker than the wheel

Jan-NiklasB, 2023-01-18:
Seems like a good idea in terms of security.
Only question is if the grid can be displayed in a readable way an if the miniBLE is capable of creating non-deterministic random numbers.

My1, 2023-02-17:
> Okay, but must be mooltipass-side only, so it's not possible to enable this feature from the multicute.

tbh I would be fine with the MP-only settings to be able to be triggered from MC but having to confirm them on the mooltipass itself so you need to navigate less menus but are still safe

CGuy-1, 2023-05-29:
A big thumbs down on this one.

The security of the Mooltipass is the selling point and it shouldn't be compromised in this way. This would only serve as an attack vector to the device. If this is what a user wants, then they should be using a S/W password manager instead.

Pushing from the device is the best security. Change the settings to reduce the number of times you need to enter a PIN if your threat model doesn't include losing physical control of the device at any time. I remove the card anytime there is a chance that I might lose physical control of the device.

The timeout can be set to 30 minutes and from the device you can disable PIN for management.

arcaartem, 2023-10-06:
> it shouldn't be compromised in this way
>This would only serve as an attack vector to the device

I don't think that's necessarily true; you're not compromising the device any more than before:

If someone can get in a position to attack while your device is connected to your PC/Phone, then you have bigger problems on your hands. They don't _need_ your PIN at that point. They don't even have to know there's a device. They can just install a malware/keylogger and steal your credentials that way.

> Change the settings to reduce the number of times you need to enter a PIN if your threat model doesn't include losing physical control of the device at any time

Interesting. So are you saying we can have options to support different threat models? 🤔

My1, 2023-10-06:
the idea is kinda interesting, although I would definitely make it a U2F-style approach where you have to confirm the pin by clicking the wheel so a piece of junkware cant go and lock up your card.

CGuy-1, 2023-10-12:
> They don't need your PIN at that point. ...
They can just install a malware/keylogger and steal your credentials that way.

With a malware/keylogger they could steal your PIN which gives them access to all your stored credentials not just the one you logged in to.

It be more secure to have the ability to type in the PIN code using a Bluetooth keyboard connected to the Mooltipass (not that Bluetooth is safe). There might be one way this could be safe is if the random starting pin was enabled an the Mooltipass only accepted scrolling and arrow keys from the Bluetooth keyboard. 

The PIN needs to have the most protection so protecting it from being intercepted is critical to the protection for all of your credentials.

My threat model is limited to a compromised computer/device and/or sites. With the Mooltipass generating random passwords for each site and choosing which device I use to access critical sites/applications I feel comfortable. If I login to some unimportant web site on a compromised device, well it isn't going to impact other sites. If I get my PIN compromised, then I'm in trouble.

My1, 2023-10-12:
> With a malware/keylogger they could steal your PIN which gives them access to all your stored credentials not just the one you logged in to.

not quite that simple. entering management needs a wheel press on the MP.

equally submitting the pin should require a wheel press anyway in my opinion.

also moving the MP to the side for a half second, the current state of the art authentication method, FIDO2, when not being used with biometrics relies also on the PIN entered on the PC, with a test of user pressence (like the touch/press of a button).

also if you wouldnt want to use the feature, dont enable it, similar to BT

arcaartem, 2023-10-12:
> With a malware/keylogger they could steal your PIN which gives them access to all your stored credentials not just the one you logged in to.

You missed my point. Once you have malware on your system, doesn't matter if your PIN can be intercepted or not, it's game over! They have full access to your system at that point, they can read the disk & the memory, intercept network, access session or login token cookies, extract personal information...etc, all of which can lead to attacks that doesn't require credentials for your other sites. It's the worst case scenario. 

> If I get my PIN compromised, then I'm in trouble.

That's assuming you don't have MFA on your accounts. Then of course, that is a problem! In which case, maybe don't enable "PIN entry through host" setting on your device? 🤷 

Perhaps I need to emphasise that I'm **not** proposing a permanent change to PIN entry mechanism but make it an optional way to enter the PIN so people with different threat models have the option to trade off some security for convenience, similar to your suggestion earlier of reducing the number of times you need to enter a PIN; it's to accommodate a different threat mode.

I'd even go so far to argue this could make the device more accessible to some people.

arcaartem, 2023-10-12:
> equally submitting the pin should require a wheel press anyway in my opinion

Or a knock! Regardless, I think that's a great idea.

CGuy-1, 2023-10-13:
> Once you have malware on your system, doesn't matter if your PIN can be intercepted or not, it's game over!

That is only the case if the malware is on your own computer/device. If my PIN is compromised then all my passwords could be compromised. My TFA is my MP, and all this is protected by a card and a PIN. It is this H/W that gives me the extra protection that I can't get with a S/W solution. I never want to have my PIN entered from an external device.

I suggested in another issue that one solution would be to keep the PIN for the first time the card is inserted and then allow a less secure method until the card is removed. Wakeup/authentication could be from a passkey, smart watch, Yubikey, NFC, etc. which would address the ease of use without having to reveal your PIN.

arcaartem, 2023-10-13:
> That is only the case if the malware is on your own computer/device. 

That's not true, that is not the only case.

>  If my PIN is compromised then all my passwords could be compromised.

The attackers would need access to your device for that to happen, no? Regardless of this feature being active, if they get your PIN (there are other ways to intercept if someone is motivated enough)

>  I never want to have my PIN entered from an external device.

I don't understand your point of view, don't enable it on your device, so no chance of that happening 🤷