mooltipass/minible #350: Can't update from bundle 4 to bundle 8

oliverpool, 2022-08-17:
_@limpkin to prevent polluting #349_

### Expected behavior

Use moolticute to update my miniBLE to bundle 8.

### Actual behavior

An error message is displayed: `Upload bundle finished with error.`

### Step by step guide to reproduce the problem

I follow the guide: https://mooltipass.github.io/minible/updating_your_device

1. disconnect miniBLE
2. insert card the wrong ways
3. connect miniBLE (`Please Remove The Card` is shown, as expected)
4. launch moolticute
5. switch to dev tab (ctrl+shift+F3)
6. Select Bundle File `update_1791_from_bundle_4_to_8_******.img`
7. Error message `Upload bundle finished with error.`

### Moolticute log

```
... startup logs
DEBUG: (2022-08-17T18:05:48.175) :0 - New message:  {"data":{"only_check":true},"msg":"request_keyboard_layout"}
INFO: (2022-08-17T18:05:55.599) :0 - Checking software updates from  "https://api.github.com/repos/mooltipass/moolticute/releases"
WARNING: (2022-08-17T18:05:56.067) :0 - Automatic updates are not suppurted in this platform  "git"
DEBUG: (2022-08-17T18:06:49.080) :0 - New message:  {"data":{"success":false},"msg":"upload_bundle"}
```

If I put the card the right way, unlock it and then insert it the wrong way, I get the following logs:
```
DEBUG: (2022-08-17T18:08:51.657) :0 - New message:  {"msg":"mp_disconnected"}
DEBUG: (2022-08-17T18:08:51.665) :0 - Setup db tracking
DEBUG: (2022-08-17T18:09:14.920) :0 - Ble Dev Tab is only available for BLE device.
DEBUG: (2022-08-17T18:09:16.319) :0 - Ble Dev Tab is only available for BLE device.
DEBUG: (2022-08-17T18:09:31.723) :0 - Ble Dev Tab is only available for BLE device.
DEBUG: (2022-08-17T18:10:06.411) :0 - New message:  {"data":{"aux_mcu_version":"","bundle_version":0,"flash_size":0,"hw_serial":0,"hw_version":"ble","main_mcu_version":""},"msg":"version_changed"}
DEBUG: (2022-08-17T18:10:06.411) :0 - Setup db tracking
DEBUG: (2022-08-17T18:10:06.411) :0 - Setup db tracking
DEBUG: (2022-08-17T18:10:06.411) :0 - New message:  {"msg":"mp_connected"}
DEBUG: (2022-08-17T18:10:06.417) :0 - New message:  {"data":false,"msg":"is_connected_with_bluetooth"}
DEBUG: (2022-08-17T18:10:06.418) :0 - New message:  {"data":{"advanced_menu":false,"bluetooth_enabled":false,"knock_disabled":false,"login_prompt":false,"pin_for_mmm":false,"storage_prompt":false},"msg":"send_user_settings"}
DEBUG: (2022-08-17T18:10:07.289) :0 - New message:  {"data":{"battery":0},"msg":"send_battery"}
DEBUG: (2022-08-17T18:10:07.289) :0 - New message:  {"data":"NoCardInserted","msg":"status_changed"}
DEBUG: (2022-08-17T18:10:15.186) :0 - New message:  {"data":{"aux_mcu_version":"0.65","bundle_version":0,"flash_size":0,"hw_serial":1791,"hw_version":"ble","main_mcu_version":"0.76"},"msg":"version_changed"}
DEBUG: (2022-08-17T18:10:15.186) :0 - Setup db tracking
DEBUG: (2022-08-17T18:10:15.187) :0 - New message:  {"data":{"aux_mcu_version":"0.65","bundle_version":0,"flash_size":8,"hw_serial":1791,"hw_version":"ble","main_mcu_version":"0.76"},"msg":"version_changed"}
DEBUG: (2022-08-17T18:10:15.187) :0 - New message:  {"data":{"aux_mcu_version":"0.65","bundle_version":4,"flash_size":8,"hw_serial":1791,"hw_version":"ble","main_mcu_version":"0.76"},"msg":"version_changed"}
DEBUG: (2022-08-17T18:10:15.193) :0 - New message:  {"data":{"battery":90},"msg":"send_battery"}
DEBUG: (2022-08-17T18:10:15.201) :0 - New message:  {"data":"Locked","msg":"status_changed"}
DEBUG: (2022-08-17T18:10:15.213) :0 - New message:  {"data":{"error_message":"Get user categories failed","failed":true},"msg":"get_user_categories"}
DEBUG: (2022-08-17T18:10:15.214) :0 - New message:  {"data":{"only_check":true},"msg":"request_keyboard_layout"}
DEBUG: (2022-08-17T18:10:15.214) :0 - New message:  {"data":{"parameter":"user_interaction_timeout","value":15},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.214) :0 - New message:  {"data":{"parameter":"random_starting_pin","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.215) :0 - New message:  {"data":{"parameter":"prompt_animation","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.215) :0 - New message:  {"data":{"parameter":"key_after_login","value":9},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.215) :0 - New message:  {"data":{"parameter":"key_after_pass","value":10},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.215) :0 - New message:  {"data":{"parameter":"delay_after_key","value":20},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.215) :0 - New message:  {"data":{"parameter":"bool_animation","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.216) :0 - New message:  {"data":{"parameter":"device_lock_usb_disc","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.216) :0 - New message:  {"data":{"parameter":"knock_sensitivity","value":1},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.216) :0 - New message:  {"data":{"parameter":"lock_unlock_mode","value":1},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.217) :0 - New message:  {"data":{"parameter":"switch_off_after_usb_disc","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.217) :0 - New message:  {"data":{"parameter":"information_time_delay","value":30},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.217) :0 - New message:  {"data":{"parameter":"keyboard_usb_layout","value":0},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.217) :0 - New message:  {"data":{"parameter":"keyboard_bt_layout","value":0},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.218) :0 - New message:  {"data":{"parameter":"device_language","value":0},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.218) :0 - New message:  {"data":{"parameter":"user_language","value":0},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.218) :0 - New message:  {"data":{"parameter":"reserved_ble","value":false},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.218) :0 - New message:  {"data":{"parameter":"prompt_animation","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.218) :0 - New message:  {"data":{"parameter":"bool_animation","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.219) :0 - New message:  {"data":{"parameter":"device_lock_usb_disc","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.219) :0 - New message:  {"data":{"parameter":"pin_shown_on_back","value":false},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.219) :0 - New message:  {"data":{"parameter":"pin_show_on_entry","value":false},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.219) :0 - New message:  {"data":{"parameter":"disable_ble_on_card_remove","value":false},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.219) :0 - New message:  {"data":{"parameter":"disable_ble_on_lock","value":false},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.219) :0 - New message:  {"data":{"parameter":"nb_20mins_ticks_for_lock","value":0},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.220) :0 - New message:  {"data":{"parameter":"switch_off_after_usb_disc","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.220) :0 - New message:  {"data":{"parameter":"information_time_delay","value":30},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.220) :0 - New message:  {"data":{"parameter":"bluetooth_shortcuts","value":false},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.220) :0 - New message:  {"data":{"parameter":"screen_saver_id","value":0},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.220) :0 - New message:  {"data":{"parameter":"key_after_login","value":9},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.221) :0 - New message:  {"data":{"parameter":"key_after_pass","value":10},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.221) :0 - New message:  {"data":{"parameter":"delay_after_key","value":20},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.221) :0 - New message:  {"data":{"parameter":"tutorial_enabled","value":false},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.221) :0 - New message:  {"data":{"parameter":"user_interaction_timeout","value":15},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.221) :0 - New message:  {"data":{"parameter":"random_starting_pin","value":true},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.221) :0 - New message:  {"data":{"parameter":"knock_sensitivity","value":1},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.222) :0 - New message:  {"data":{"parameter":"lock_unlock_mode","value":1},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:15.222) :0 - New message:  {"data":{"parameter":"hash_display","value":false},"msg":"param_changed"}
DEBUG: (2022-08-17T18:10:17.321) :0 - New message:  {"data":{"success":false},"msg":"upload_bundle"}
```

### Firmware Version
AUX MCU version:  0.65
Main MCU version:  0.76
Bundle version:  4

### Moolticute Version - If Involved

v0.55.0

### Operating System

archlinux

limpkin, 2022-08-17:
does the device freeze for a few seconds before you get the error message?

oliverpool, 2022-08-18:
I get the `Upload bundle finished with error.` almost instantly after selecting the file

oliverpool, 2022-08-19:
yes, always on arch linux

oliverpool, 2022-08-19:
I was wondering which debugs logs you wanted (I thought that what moolticute sent was already a debug log).

I looked up the manual and saw the checkbox in the settings, however the "View Logs" windows was mainly empty.

I finally realized that `moolticuted` was being run by systemd. I stopped the service and let it run manually `moolticuted -l` and everything worked fine!

I looked at the systemd logs `journalctl -b -u moolticuted.service` and saw the offending line:
```
Aug 19 15:36:37 oliskull moolticuted[533]: DEBUG: (2022-08-19T15:36:37.351) :0 - JSON API recv: {
Aug 19 15:36:37 oliskull moolticuted[533]:     "data": {
Aug 19 15:36:37 oliskull moolticuted[533]:         "file": "/home/.../minible/update_1791_from_bundle_4_to_8_****.img",
Aug 19 15:36:37 oliskull moolticuted[533]:         "password":"<masked>"
Aug 19 15:36:37 oliskull moolticuted[533]:     },
Aug 19 15:36:37 oliskull moolticuted[533]:     "msg": "upload_bundle"
Aug 19 15:36:37 oliskull moolticuted[533]: }
Aug 19 15:36:37 oliskull moolticuted[533]: CRITICAL: (2022-08-19T15:36:37.351) :0 - Error opening bundle file:  "/home/.../minible/update_1791_from_bundle_4_to_8_***>
Aug 19 15:36:37 oliskull moolticuted[533]: CRITICAL: (2022-08-19T15:36:37.351) :0 - "Bundle file is not readable"
```

So `moolticuted` had simply no right to read the bundle file because of the [systemd config](https://github.com/mooltipass/moolticute/blob/master/systemd/moolticuted.service).

---

I looked at editing the service by adding the `User=<myself>` and `Group=plugdev`, however the `plugdev` group does not exist on arch linux.
I created the group `sudo groupadd plugdev` and added a [udev rule](https://unix.stackexchange.com/a/85459):
```
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", MODE="0660", GROUP="plugdev"
```

Now the daemon is running as my user and I will try to update my other device as soon as the battery has enough power!

---

I am closing this issue, since my problem is solved. However I see two improvements that could be made to address such issues:
- have a more specific error message  (`Bundle file is not readable` for instance) instead of the generic `Upload bundle finished with error.`
- forward the file content to the daemon instead of giving the filepath to the daemon and letting it read the file

oliverpool, 2022-08-19:
@limpkin but as alwaysm thank you very much for the prompt help!

closed 2022-08-19 by oliverpool

limpkin, 2022-08-19:
awesome! glat it's working now :)
tagging @deXol to tackle this in the future :)

oliverpool, 2022-08-19:
With the config changes (`moolticuted.service` and `99-hidraw-permissions.rules`), I was able to update my other device :tada: