mooltipass/minible #352: Not working over Bluetooth on Linux

hijackeel, 2022-09-15:
### Expected behavior
Over Bluetooth, I should be able to enter credentials manually, use the automatic Firefox extension, and use Moolticute.

### Actual behavior
I can pair my Mooltipass with my Linux computer (but it seems quite finicky - sometimes failing to pair, and sometimes rapidly connecting/disconnecting, or both).
However, once successfully paired, nothing works anyway.
Everything works over a USB cable.

### Step by step guide to reproduce the problem
1. Pair Mooltipass with Linux computer.
2. Try manually entering credentials into a text field (Mooltipass seems to send credentials, but nothing shows up on computer).
3. Open Moolticute (it says "Connect Your Mooltipass Device").

### Firmware Version
AUX MCU version:  0.70
Main MCU version:  0.80
Bundle version:  8

### Moolticute Version - If Involved
Latest Git commit: 0331c6d59098aa91fe9236ebbf85c1319c3b7ad7

### Operating System + other software
Gentoo Linux
bluez 5.65 (so seemingly unrelated to https://github.com/mooltipass/moolticute/issues/993)
blueman-applet 2.3.2

hijackeel, 2022-09-15:
By the way, a Bluetooth speaker I have does work.

limpkin, 2022-09-15:
Before we start investigating more in details, just to make sure, could you update to a more recent bluez? :)

hijackeel, 2022-09-15:
I'm on 5.65, the latest.

limpkin, 2022-09-15:
could you build bluez from source?

hijackeel, 2022-09-15:
I did; my distro is Gentoo.

limpkin, 2022-09-15:
my bad...
so if I understand correctly your report, neither manual credential recall (having the Mini BLE type characters) nor communications work with moolticute?

hijackeel, 2022-09-15:
Correct. Nothing works once paired. It's also difficult to pair it. I tried again just now, and the Mooltipass shows the code entry screen, but my computer doesn't show the code to enter, instead acting like the Mooltipass is already successfully connected. The GUI I'm using to do this is blueman-applet in swaybar.

limpkin, 2022-09-15:
@hijackeel would you have access to a non linux computer to check if it's a device or OS related issue?

hijackeel, 2022-09-17:
@limpkin It works over Bluetooth with Windows 10 and Android.

limpkin, 2022-09-19:
uh that doesn't sound good.
@principis @barathrm : is everything still working ok on your side?

principis, 2022-09-19:
Works fine for me (also Fedora)! Currently quite time limited, I'll check back when I have more time.

barathrm, 2022-09-19:
Blue 5.65 on openSUSE working fine here. Manual recall works. I'd check if there are any configure flags used when building bluez for gentoo which disable some feature required for this to work... I'd debug by trying to connect to the minible after having started bluetoothd manually. You can follow the procedure to run `bluetoothd` and and connecting via `bluetoothctl` manually here in this old issue ticket https://github.com/mooltipass/moolticute/issues/671#issue-621211433

Only caveat is that I need an extr udev rule for moolticute to detect it via BT (which I've still forgotten to make PR for adding by default and maybe digging into why it's become necessary lately) https://github.com/mooltipass/moolticute/issues/993#issuecomment-1103173135.

hijackeel, 2022-09-22:
I don't appear to have any relevant use flags disabled.

When pairing manually via bluetoothctl, the following shows up in the systemd log (debug enabled):
> Sep 22 11:51:13 kotir bluetoothd[8630]: src/device.c:btd_device_set_temporary() temporary 0
Sep 22 11:51:13 kotir bluetoothd[8630]: src/agent.c:agent_ref() 0x564d5eb16fb0: ref=2
Sep 22 11:51:13 kotir bluetoothd[8630]: src/device.c:bonding_request_new() Requesting bonding for 68:79:12:30:09:00
Sep 22 11:51:13 kotir bluetoothd[8630]: src/agent.c:agent_ref() 0x564d5eb16fb0: ref=3
Sep 22 11:51:13 kotir bluetoothd[8630]: src/agent.c:agent_unref() 0x564d5eb16fb0: ref=2
Sep 22 11:51:13 kotir bluetoothd[8630]: src/adapter.c:suspend_discovery()
Sep 22 11:51:13 kotir bluetoothd[8630]: src/adapter.c:adapter_bonding_attempt() hci0 bdaddr 68:79:12:30:09:00 type 1 io_cap 0x04
Sep 22 11:51:13 kotir bluetoothd[8630]: src/shared/mgmt.c:send_request() [0x0000] command 0x0019
Sep 22 11:51:13 kotir bluetoothd[8630]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000b
Sep 22 11:51:13 kotir bluetoothd[8630]: src/adapter.c:connected_callback() hci0 device 68:79:12:30:09:00 connected eir_len 30
Sep 22 11:51:13 kotir bluetoothd[8630]: src/gatt-database.c:connect_cb() New incoming LE ATT connection
Sep 22 11:51:13 kotir bluetoothd[8630]: attrib/gattrib.c:g_attrib_ref() 0x564d5eafee10: g_attrib_ref=1
Sep 22 11:51:13 kotir bluetoothd[8630]: src/device.c:load_gatt_db() Restoring 68:79:12:30:09:00 gatt database from file
Sep 22 11:51:13 kotir bluetoothd[8630]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:exchange_mtu_cb() MTU exchange complete, with MTU: 5
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0017
Sep 22 11:51:14 kotir bluetoothd[8630]: src/adapter.c:user_passkey_notify_callback() hci0 68:79:12:30:09:00
Sep 22 11:51:14 kotir bluetoothd[8630]: src/adapter.c:user_passkey_notify_callback() passkey 284887 entered 0
Sep 22 11:51:14 kotir bluetoothd[8630]: src/device.c:new_auth() Requesting agent authentication for 68:79:12:30:09:00
Sep 22 11:51:14 kotir bluetoothd[8630]: src/agent.c:agent_ref() 0x564d5eb16fb0: ref=3
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_primary_cb() Primary services found: 7
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discovery_parse_services() start: 0x0001, end: 0x000
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discovery_parse_services() start: 0x0008, end: 0x000
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discovery_parse_services() start: 0x000c, end: 0x001
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discovery_parse_services() start: 0x0019, end: 0x001
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discovery_parse_services() start: 0x001d, end: 0x003
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discovery_parse_services() start: 0x0036, end: 0x004
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discovery_parse_services() start: 0x0044, end: 0x004
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() Characteristics found: 27
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0002, end: 0x0003, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0004, end: 0x0005, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0006, end: 0x0008, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0009, end: 0x000c, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x000d, end: 0x000e, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x000f, end: 0x0010, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0011, end: 0x0012, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0013, end: 0x0014, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0015, end: 0x0016, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0017, end: 0x0019, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x001a, end: 0x001d, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x001e, end: 0x001f, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0020, end: 0x0022, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0023, end: 0x0025, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0026, end: 0x0027, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x0028, end: 0x0029, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x002a, end: 0x002d, valu
Sep 22 11:51:14 kotir bluetoothd[8630]: src/shared/gatt-client.c:discover_chrcs_cb() start: 0x002e, end: 0x0031, valu
[... cut ...]
Sep 22 11:52:03 kotir bluetoothd[8630]: src/gatt-client.c:btd_gatt_client_ready() Update Features 0x00
Sep 22 11:52:03 kotir bluetoothd[8630]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_68_79_12_30_09_00 err 0
Sep 22 11:52:03 kotir bluetoothd[8630]: src/shared/gatt-client.c:service_changed_register_cb() Registered handler for

This is the complete contents of my mooltipass rule file:
> SUBSYSTEM=="usb", ATTRS{idVendor}=="16d0", ATTRS{idProduct}=="09a0", TAG+="uaccess"

I tried editing as suggested, but it didn't work.

limpkin, 2022-11-05:
@hijackeel any news on this front? 
We also recently released beta bundle v12 at beta-updates.themooltipass.com which might help in your case.