mooltipass/minible #359: Autotype of credentials without permission

Jan-NiklasB, 2022-11-13:
### Expected behavior
Working on several websites the mooltipass detects credential fields and asks before entering credentials

### Actual behavior
The mooltipass enters credentials without permission if unlocked

### Step by step guide to reproduce the problem
Can't be given here, because most of the sites with this behavior are confidential.
(Based on Portainer CE or own web development of the company)

In Portainer this problem shows if on enables the password protection of the backup, in this moment the mooltipass enters the password saved for the user in this field

### Firmware Version
AUX MCU version:  0.70
Main MCU version:  0.80
Bundle version:  8

### Moolticute Version - If Involved
0.55.0

### Operating System
Mention if you are using either:
platform independent (tested on Win7, Win 10 and Linux Mint
 

### Mooltipass Extension
- If you're creating an issue to report a website incompatibility, please use the "Report incompatibilities with this website" item in the extension menu.
- If you're creating an issue to report a problem with our extension, please create another issue here: 
https://github.com/mooltipass/extension/issues

-> Read this, but since the mooltipass sends the credentials without asking, I guess it's a firmware problem

limpkin, 2022-11-13:
Hello there!
Without having a step by step guide this is going to be quite hard for us to debug this issue... is it possible that you're encoutering a side effect of our credential cache? For a given website & tab, if 2 credentials requests happen within 5 seconds of each other than the second one will automatically be approved.

Jan-NiklasB, 2022-11-13:
Nope, I was logged in on Portainer for at least 15 min. and then created a backup which I wanted to be pw-protected.
So on enableing the pw option it entered the credentials.

I would be fine with showing this example in a anydesk session.
Also this should be the same on any portainer installation... the other side I can't demonstrate, because it contains customer data (customer administration and also our support website running on zammad)

limpkin, 2022-11-27:
I just thought of something... could you maybe disable and re-enable the extension when you're logged in? I'd be curious to see if you get another credential request the moment you re-enable it :)

Jan-NiklasB, 2022-12-04:
Yep, got annother request....

Jan-NiklasB, 2022-12-04:
BTW, this is what it looks like, I only log out, everything else happens without touching anything:

https://user-images.githubusercontent.com/44967908/205493922-55517196-925c-4921-b819-9c478c768be4.mp4

ananthb, 2022-12-04:
The mooltipass logs me in automatically without confirmation too. I thought this was the expected behaviour in simple mode. Maybe i read this in the manual. I think it mentioned that confirmation is requested only in advanced mode.

limpkin, 2022-12-04:
I wonder if this is due to the fact that with this website, the extension is somehow always submitting the password, and therefore the 5seconds buffer never times-out... 
flagging @extensionssoft