mooltipass/minible #361: Moolticute not quitting properly on macOS Monterey

Lhudwolf, 2022-11-29:
### Expected behavior

When i quit the program (CMD + Q) the process is killed. I can reopen Moolticute on macOS without using the terminal to manually kill the process.

### Actual behavior

I quit the program and Moolticute is not completly closed. I cannot reopen the program without firstly going into the terminal, pgrep moolticute and killing the provided PID manually.

### Step by step guide to reproduce the problem

1. Open Moolticute on macOS
2. Quit Moolticute on macOS
3. Try to reopen Moolticute on macOS

### Moolticute Version

v1.00.0

### Operating System

MacOS Monterey 12.6.1

limpkin, 2022-11-29:
does right clicking on the tray icon and then selecting "quit" work?

Lhudwolf, 2022-11-29:
> does right clicking on the tray icon and then selecting "quit" work?

No, unfortunately not. Same result if i try to quit it through the top bar context menu ("Datei" --> "Beenden").

This is not a new issue - i had it with previous versions as well. I can confirm that the bug is also present on a friends MacBook Air M1.

Unrelated question - is providing a Big Sur sized App icon for the Mac version planned?