mooltipass/minible #365: Field for additional information

EugeneRymarev, 2022-12-16:
### Missing function
Additional field for account comment.

### Justification
Some sites require comments to be saved, such as account recovery codes, 2FA rescue codes, or something else.

### Workarounds
Creating Notes is inconvenient, since it is a different entity and needs to be looked for in a completely different place.

barathrm, 2022-12-16:
We already have a description field. Based on https://github.com/mooltipass/minible/wiki/Mooltipass-Protocol#0x0006-store-credential I can't immediately see if there's a size limit, but moolticute restricts it to something small and single line anyways.

If the description field can't be used, maybe a credential could be linked to a note?

EugeneRymarev, 2022-12-16:
@barathrm, it is not comfortable. Multiline notes like these look better.

[image]