mooltipass/minible #366: Steam TOTP

EugeneRymarev, 2022-12-25:
### Missing feature
Steam TOTP

### Justification
Why not? 😊

### Workarounds
Using WinAuth or Steam Android App 👎

[Python implementation](https://github.com/Elipzer/SteamAuthenticatorPython/blob/master/main.py)

EugeneRymarev, 2022-12-26:
```python
import hmac
import hashlib
import time
import base64
from binascii import unhexlify

def generate_steam_totp(secret_key):
    steam_chars = '23456789BCDFGHJKMNPQRTVWXY'
    len_steam_chars = len(steam_chars)
    v = unhexlify('%%0%dx' % (64 // 4) % int(round(time.mktime(time.localtime(time.time())) * 1000) / 30000))
    digest = hmac.new(base64.b32decode(secret_key), v, hashlib.sha1).digest()
    start = int(digest[19].encode('hex'), 16) & 0x0f
    full_code = int(digest[start:start + 4].encode('hex'), 16) & 0x7fffffff
    code = ''
    for _ in range(5):
        code += steam_chars[full_code % len_steam_chars]
        full_code /= len_steam_chars
    return code
```

Jan-NiklasB, 2022-12-28:
@EugeneRymarev Where do you get the TOTP-Key from?
I tried this a while before but couldn't find it anywhere...

EugeneRymarev, 2022-12-28:
> Where do you get the TOTP-Key from?

@Jan-NiklasB, you can get it from rooted Android device or rooted WSA.

Jan-NiklasB, 2022-12-28:
Ah okay, so no 'legit' way to get hold on it...
And I thought I mieses something. ..

But I will try of I can emulate a rooted android device and extract the key, maybe someone here can use this info

Ancyker, 2024-09-21:
Just commenting to add that YubiKey/Yubico Authenticator _does_ support Steam's OTPs as an undocumented feature. Starting an account name with `Steam:` (i.e. `Steam:exampleuser`) enables the feature.

[image]