mooltipass/minible #368: Steam confimations

EugeneRymarev, 2023-01-03:
### Missing feature
Along with [Steam TOTP](https://github.com/mooltipass/minible/issues/366), I would also like to add trade/sale confirmations for Steam

### Justification
To be a complete replacement for Steam Authenticator/[WinAuth](https://github.com/winauth/winauth).

### Workarounds
Use Steam Authenticator/WinAuth.

My1, 2023-02-17:
just for completeness sake, aside from the fact there's no legit way to obtain the secrets needed (as mentioned in the Steam TOTP issue), this one also needs to have internet access and stuff because the confirmation actually contains trade data.

Aside from that Steam seems to be phasing out the Steam TOTP in favor of the QR codes which are much harder to phish (not killing yet, however the TOTPs clearly seem to be more hidden)

Frankly, if MP supports txAuth steam could go with FIDO for this lol

EugeneRymarev, 2023-02-17:
@My1, I doubt that someday Steam will abandon TOTP. And even if something like this happens, it means absolutely nothing, since there will be a ready-made code, which, in the future, will be needed for other similar authorization systems.

My1, 2023-02-18:
but even then there isnt a legit way to obtain the secrets, and unlike WinAuth which is "just" an open source project, mooltipass is an actual product that's actually sold. valve might not look overly kindly on that.

EugeneRymarev, 2023-02-19:
@My1, You can not write that this is Steam. Write something abstract.

Actually, I disagree with you that there is no legal way to get this code. The Android system is an open source system. And the Steam app for Android allows you to install yourself on any Android system, including rooted ones. Therefore, I do not consider this method of obtaining the key "illegal", as you call it.

My1, 2023-02-19:
I said legit, not legal.
I am not saying that it's illegal to obtain the code, however you aren*t supposed to get the code that way.
also considering that steam is massive a threat with lawyers doesnt even have to be about something illegal exactly, MP likely wouldnt have the resources to fight such things.

EugeneRymarev, 2023-02-19:
@My1, Fine. How do you think WinAuth managed to create an algorithm for generating codes for Steam and confirming transactions for the exchange and sale of items, if Steam themselves did not provide information to perform these actions?

If indeed there is information somewhere that Steam prohibits the creation of something like this, then provide me with sources. I did not find such information.

My1, 2023-02-19:
I am just saying that it's pretty sure you are kinda not supposed to do it, as after all you can only obtain the relevant data by going through protections the app has from the OS and you effectively backdooring your account.

> How do you think WinAuth managed to create an algorithm for generating codes for Steam and confirming transactions for the exchange and sale of items, if Steam themselves did not provide information to perform these actions?

reverse engineering obviously, doesnt have to be by the winauth steam themselves, but i'd assume someone reversed it, wrote something and WinAuth wrote something based on that.

their API docs don't seem to have anything regarding Steam Guard at least
https://developer.valvesoftware.com/wiki/Steam_Web_API

EugeneRymarev, 2023-02-19:
@My1, The WinAuth project is more than 10 years old - during this time, Steam has no complaints either against it or any other similar projects. Also, I have not noticed that Steam has ever changed its algorithm for two-factor authentication.

From the above, I conclude that Steam will not make any claims about this implementation in Mooltipass.