mooltipass/minible #369: Long passwords are entered with spaces that are not in these passwords

EugeneRymarev, 2023-01-06:
### Expected behavior
Do not enter characters that do not exist
[image]

### Actual behavior
[image]

### Step by step guide to reproduce the problem
Manual selection and entry of credentials.
Password length is 32 characters.
A space is always after the 14th character.

### Firmware Version
AUX MCU version:  0.73
Main MCU version:  0.84
Bundle version:  12

### Moolticute Version
v1.00.1

### Operating System
- Windows 11 Pro (Russian language is main)

limpkin, 2023-01-06:
This seems to be a dead key issue: on standard keyboard layout, I can't type '^' by itself, I need to type '^' and then add a space. The Mini BLE therefore does the same. Could you let us know which layout is set on your computer when you experience that issue?

EugeneRymarev, 2023-01-06:
> Could you let us know which layout is set on your computer when you experience that issue?

@limpkin, ENG
[image]

If I forgot to change the layout, I see this
[image]
[image]

limpkin, 2023-01-06:
I'm guessing this is only happening with the ^ character, but I'm not sure why a windows system is affected.
Can you confirm that when you type '^' on your keyboard it gets directly typed?

barathrm, 2023-01-15:
> I'm guessing this is only happening with the ^ character, but I'm not sure why a windows system is affected. Can you confirm that when you type '^' on your keyboard it gets directly typed?

I guess the issue here is that the minible can't be configured with a russian layout. so it sends characters as if it was an english (?) layout, which is interpreted incorrectly by the operating system expecting a russian keyboard (and the corresponding key events).

EugeneRymarev, 2023-01-16:
> english (?) layout

@barathrm, i have 2 layouts
1. on ENG layout i receive `^ ` with space
2. on RUS layout i receive `: ` with space

barathrm, 2023-01-16:
> > english (?) layout
> 
> @barathrm, i have 2 layouts
> 
>     1. on ENG layout i receive `^ ` with space
> 
>     2. on RUS layout i receive `: ` with space

Yeah, but I mean on the mooltipass. The mooltipass itself is also configured with a keyboard layout to "pretend" to be a keyboard of that type and that a human presses keys on that type of layout.

I mean here, in moolticute, there is no russian layout, so it will have to pretend to be some other layout for now (and the OS layout must match):

[image]

EugeneRymarev, 2023-01-16:
> Yeah, but I mean on the mooltipass. The mooltipass itself is also configured with a keyboard layout to "pretend" to be a keyboard of that type and that a human presses keys on that type of layout.

@barathrm, So you need to add the Russian language.

To be honest, I still do not understand what the problem is to add it at least somehow. There are a lot of ready-made fonts with the Russian language. If the problem is in the translation, then I can help.

limpkin, 2023-01-16:
@barathrm thanks for the help!
What's odd to me is that even if @EugeneRymarev layout is configured as english, that the deadkeys behavior isn't the same as an english layout... why for the same layout the Mini shouldn't send the space?

barathrm, 2023-01-16:
> @barathrm thanks for the help! What's odd to me is that even if @EugeneRymarev layout is configured as english, that the deadkeys behavior isn't the same as an english layout... why for the same layout the Mini shouldn't send the space?

I think it's because on many layouts, you get a `^` by press `Shift, ^, Space`. But on a russian keyboard, the ^ might have its own dedicated key, so a dead-key is not required to produce it. In that case, the fix would be to send only `Shift, ^` (or 6, if that's the keycode... I always forget/get confused by keycodes vs scan codes etc)

limpkin, 2023-01-16:
on the french layout the ^ also gets his own key... yet i need to send the space key too :D

limpkin, 2023-01-16:
you're correct @barathrm

limpkin, 2023-01-17:
@EugeneRymarev is there any character that you can only type if it is followed by a space bar press?

EugeneRymarev, 2023-01-18:
@limpkin, no

limpkin, 2023-01-18:
how can you type â then?

EugeneRymarev, 2023-01-18:
> how can you type â then?

@limpkin, i don't have this symbol in my layout RUS.
That is French language - i know only `Alt + NumPad 0226` for this symbol

limpkin, 2023-01-18:
hmm... so what's the use of ^ for your english layout in that case, if you can only type it and can't combine it with anything else?

EugeneRymarev, 2023-01-18:
@limpkin, such things are usually done by the editor in which it works, for example Word.

limpkin, 2023-01-18:
Anyway, for this bug, when testing locally I can't replicate your issue (mooltipass layout set to english, windows layout set to english): no space character is sent.
What layout do you select in moolticute settings?

EugeneRymarev, 2023-01-18:
@limpkin, nice. Now without space after `^` symbol.

closed 2023-01-18 by limpkin

My1, 2023-03-12:
> @limpkin, such things are usually done by the editor in which it works, for example Word.
> 
> It seems to me that if MiniBLE emulates keyboard input, then special characters (dead keys) should be entered through Alt codes.

no thanks, iirc that only works in windows so on Linux that would be breaking stuff needlessly

My1, 2023-07-27:
> @limpkin, nice. Now without space after `^` symbol.

makes sense, US intl has some dead keys active to specifcally be able to make international characters without needing alt codes.