mooltipass/minible #370: Changing the bluetooth connection section

EugeneRymarev, 2023-01-11:
### Missing feature
1. Select the device to be paired with from the MiniBLE menu.
2. If the selected device is not available, then go to sleep/deep sleep according to the timer.
3. MiniBLE should store a list of paired devices that can be edited (renamed, deleted) via Moolticute.
### Justification
The bluetooth connection section in the form in which it currently exists and, in general, the MiniBLE connection to managed devices itself is unstable.
What do I mean by instability? I have two devices paired with MiniBLE - a laptop and a phone. Periodically, usually in the morning when MiniBLE is turned on for the first time, the device connects to the phone, although it was previously connected to a laptop, but, apparently due to a loss of connection due to rebooting or turning the laptop off and on, the device switches to the next one it has ever been paired with and work with him.

### Workarounds
The current implementation, unfortunately, leaves much to be desired.

alanrick, 2023-01-25:
Sadly, this instability is so severe it's prevented me from using the Bluetooth capabilities for over 6 months. So many thanks @EugeneRymarev Eugene for raising the issue. 🙏

Even with a simple constellation, iPad and iPhone,  switching device very rarely succeeds. So in practice I just use the display function rekeying the password into my iOS device. As you can imagine, particularly as many sites hide the password as it's typed in, this is very awkward.

Steps to reproduce.
1. [BLE] Delete all pairings.
2. [BLE+iPhone] Pair with iPhone
3. [BLE] Disconnect device. Repeat until the message is displayed that no connections are available. 
4. [iPhone] Switch off Bluetooth  
5. [iPad] Pair with iPad.
6. [BLE] Disconnect device
7. [iPhone] Switch on Bluetooth 
8. [iPhone] In the iPhone Bluetooth settings display, press _connect_ for the BLE.

===> Sometimes the BLE does connect with the iPhone, Sometimes the iPhone displays the failed message but then the status does change to connected (timing issue?), but mostly connection simply fails and you have to keep retrying. Often during a connection attempt the sleep message on the BLE displays and you have to move the wheels to keep it awake, which may interfere with the handshaking (guessing here).

Initially I tried the workaround of deleting all pairings every time I wanted to reconnect with a different device but this is also very cumbersome as you have to use the scroll wheels to enter a 6 digit code every time.

It might help if the BLE displayed the name of the connected bluetooth device. But ultimately this device-switching has to improve to make bluetooth viable on the BLE.

EugeneRymarev, 2023-01-25:
@alanrick, Thank you for your comment! This is a really big problem if you have more than one bluetooth connected device. I really look forward to and hope that the bluetooth section will be completely redesigned.

alanrick, 2023-01-25:
It just dawned on me how inappropriate it is having a feature request rather than a bug for the inability of a device with Bluetooth in its name to connect to different devices reliably. All my other Bluetooth devices from headphones, car, locks... to key-finders can.

kbembedded, 2023-02-24:
From some testing, while not perfect, there should be no reason to delete pairings as a workaround. I agree that some kind of in-device way to specify what to pair to would be ideal, but I'm not intimately familiar enough with Bluetooth LE to know how feasible this is.

AFAICT the miniBLE will connect with the most recently connected device if its available and automatically connectable. I use it between an iPhone and a Linux laptop. The miniBLE generally prefers the iPhone. But if you go in to the Bluetooth menu on the miniBLE, you can select `Disconnect Device` at which point the miniBLE will disconnect from the currently connected device. From there, any other already paired device can connect the to miniBLE without issue.

FWIW I've never come across any Bluetooth / LE device that can correctly/easily/functionally connect to a given paired device. Mostly commonly I do it with a couple of BT audio devices and it takes minutes if not 10+ minutes to correctly get the device to re-connect. Often I *have* to delete pairings from every device it could connect to and re-pair with what I want. Bluetooth is a lawless wasteland and the miniBLE is no exception.

EugeneRymarev, 2023-02-25:
@kbembedded, an example of [device](https://www.logitech.com/en-ph/products/mice/m720-triathlon.html). This is a mouse with a physical button to switch from one device to another. I would like the same thing to be implemented in MiniBLE only at the software level. This would be much more convenient than hoping that MiniBLE will connect to the device I need.

I also found another problem. I connected my MiniBLE for charging to the laptop with which I use it via Bluetooth. After MiniBLE was charged, I disconnected it from USB and tried to use MiniBLE for its intended purpose. Unfortunately, I did not succeed until I completely removed the pairing in Windows 11 and on MiniBLE, and then reconnected them.

Issues like this with Bluetooth are very disruptive to the user experience and put off using the device.

alanrick, 2023-02-25:
@kbembedded I suspect I missed off one step in my description but I won’t be able to verify this for a few days. But I’m pretty sure I performed step 
6.5 [ipad] switch off Bluetooth

in other words I’d expect the BLE to have no active connections because Bluetooth on all remote devices is off, but it should retain an internal bookkeeping list of including both the iPad and iPhone pairing keys. 

So *step 7* should allow the BLE to check this list, discover the iPhone pairing key, and pair with the iPhone, because no other Bluetooth devices are active. This is how my other Bluetooth devices behave.  But unfortunately the BLE doesn’t pair. 

The work around is to use a USB cable for a PC, or to delete all pairings and re-pair, which is awkward as you have to enter a 6 figure code on the BLE - an unnecessary and cumbersome activity with my fat fingers.  

I’ll check next week that I really did perform step 6.5.

limpkin, 2023-02-28:
A suggestion from a customer: 

Ideally.

P0: I want to see a list of paired devices.  Name and ID some kind of friendly name and the mac address.
P1: I want to manage this list of paired devices to remove previously paired devices.
P2: I want to assign priorities of the device list (this could just be re-ordering a list) or perhaps more complicated give priority (like a router hop metric), use lower cost over higher cost.  This is to cover that you might have different scenarios with different devices.  If this is connected use it first, if this is not available this use that.
In the most basic way – this is just an orderd list.  
In fact, not even sure that auto-connecting between items in a list is desirable.
The most basic approach is , you connect a device, it stays connected until you disconnect it.  
If the connection drops, re-connect to that device.
I think it might be questionable to EVER connect to another device automatically.
If it is this a thing should be an option to disable.  NEVER connect automatically vs ALWAYS connect automatically vs ASK ME TO CONNECT.  All 3 options is the best approach, but if I can only have 1 option
 

Most basic level , re-connect to the last paired / active device, and NEVER CONNECT to a different device without notification or asking me even if it was previously paired.
Once paired it can be selected as a device to connect to.
Unless selected it should never be auto connected.
i.e. if one BT device goes away don’t automatically connect to some other device. (this could be desirable in a commuter scenario though where the device travels into different environments)
but that also needs to be through though.
 

To start
Present a list of devices.
Let user delete one specific device.
Let user pick which device connection is active.
Don’t roam or auto-connect unless that’s really been thought through with different scenarios with different environments and use cases.

EugeneRymarev, 2023-03-01:
@limpkin, I agree completely that you can not automatically switch between devices. This does not need 100% since it is not safe.
I [suggested](https://github.com/mooltipass/minible/issues/370#issuecomment-144492334) version of manual switching between devices to press the hot key (double/triple click on the wheel).
Once again I will describe my vision.
1. List of paired devices. This list can be edited via Moolticute. Two fields are enough - MAC address and name.
2. You can manually remove any of the records on minible, and not clean everything at once.
3. You can manually on the device change the controlled device (laptop, phone, PC, etc.) by selecting from the list or the carousel method to press the hot key.
4. If the controlled device does not answer, then go into sleep/deep sleep
5. When you turn on minible, try to connect only to the last controlled device, if it does not answer, then go into sleep/deep sleep, for example, in a minute.

This is a minimum implementation that is now very needed.

alanrick, 2023-03-01:
> I think it might be questionable to EVER connect to another device automatically.
I strongly disagree.

I can imagine two scenarios. 
1. _Office_ - a single device is shared by several users (e.g. PCs) . Ie. The user has no control over which devices are available (i.e. Which devices are switched on with bluetooth on within range) 
2. _Home_ - The user has several devices and full control over which device is available.

My scenario is most definitely the latter, _Home_ (also on the move). I want to connect to my iPhone or my iPad or PC.

For this reason, I most definitely prefer auto-connection, because I can simply temporarily switch off the bluetooth on devices within range that I do **NOT** want to connect to, leaving one device (e.g. iPhone). This way I control which device I want to connect to without having to make a selection on my BLE. It's a question of usability. The iOS/PC is easier to control because of screen-estate, mouse... etc.

If this is not the case, and a list is available to select from, then three fields are needed:

1. Name.
2. MAC address (but do all devices have this? I thought this was only for networking devices. not direct handshaking)
3. Auto-connect flag

This is what's usually offered for wifi connections in iOS, Android and Windows, so it's been proved in practice.

alanrick, 2023-03-01:
> I suspect I missed off one step in my description but I won’t be able to verify this for a few days. But I’m pretty sure I performed step
> **6.5 [ipad] switch off Bluetooth**

I could not reproduce this problem today. The only problem I had was that step 5 would not work unless I insert step 4.5 [BLE] Switch off Bluetooth and then switch on again.

alanrick, 2023-04-11:
Reminder to myself...
If Windows bluetooth setting shows the BLE alternating between paired and connected every 2 seconds then delete the Windows pairing and pair the devices again.

In the meantime I can toggle between BLE on my iPhone and iPad without problems. Fingers crossed it stays that way.

CGuy-1, 2023-05-29:
I agree Bluetooth needs to be able to display the "name" or ID of the currently paired device and I should be able to switch to another device from the list of paired devices.

I have had MANY occasions where I disconnect the Bluetooth device only to have it reconnect before I can connect to the device I want. Turning off Bluetooth on all devices except the one I want to connect to isn't a workable solution as this may mean disconnecting other devices as well.

Just being able to name the connected devices and SEE what I have in my list is a great start. Being able to select a device to connect to would be high on the list of features.

Perhaps asking which Bluetooth device to connect to after entering the PIN when the card was inserted would be a good compromise with auto connect from a security point. I've experienced my credentials appearing on another device because I didn't realize it connected to a different device than the one I was on.