mooltipass/minible #372: Get TOTP Code protocol message returns invalid codes

barathrm, 2023-01-24:
### Expected behavior

The TOTP codes you get from this message should be valid and the same as when you request a code manually from the minible via the scroll wheel.

### Actual behavior

The get TOTP code message returns codes which don't work. It's a bit tricky to do, but the OTP codes you get from this message don't seem to be the same as the codes you get by manually requesting TOTP codes from the minible.

### Step by step guide to reproduce the problem

Use moolticuted + json WS requests to ask for TOTP tokens for a service. Test the token.

One can also enable `Display TOTP After Login Into Website`, then requesting a login. You'll see how long a token is valid, so you can time it to request a token via the new protocol message within that time. The tokens seem to be different.

### Firmware Version
AUX MCU version:  0.73
Main MCU version:  0.84
Bundle version:  12

### Moolticute Version - If Involved'

1.00.1, but also tested manually sending the request via usb.

### Operating System
Mention if you are using either:

Linux

I'm attaching a simple python script to test TOTP with. It connets to moolticuted, asks for a TOTP token and then a credential. You can run it with `./wstest.py <service> <login>`. 
[wstest.py.txt](https://github.com/mooltipass/minible/files/10494293/wstest.py.txt)

limpkin, 2023-01-24:
thanks for testing that @barathrm ! i'll give it a look

My1, 2023-02-17:
I think one issue might be time, is there a way to display the current time the Mooltipass has on file?

barathrm, 2023-02-17:
I don't think the internal time is an issue here, since OTP works when requesting an OTP code manually via the minible's scroll wheel and display. It just doesn't work when requesting an OTP code via the HID protocol message https://github.com/mooltipass/minible/wiki/Mooltipass-Protocol#0x0041-get-totp-code. Note that this message isn't actually used by moolticute or the browser extension yet (last I checked), I tested this manually with my own application.

My1, 2023-02-17:
one thing that might be worth checking is if the index is borked somewhere on the way.

e.g. 
1) make a clean db with a few entries and get the TOTPs via HID, important: note the times when obtained.
2) make the TOTPs manually with the same params, and see if they are perhaps moved

you can equally try this on the database you already have but moolticute doesnt let you export your seeds (usually it isnt a good idea anyway)

limpkin, 2023-02-17:
sorry I should have gotten to this sooner... will try to tackle it next week :/

limpkin, 2023-08-28:
I can't believe it took me that long :/

closed 2023-08-28 by limpkin

barathrm, 2023-08-29:
> I can't believe it took me that long :/

Don't worry about that, we all have lives! :) And thank you! Can't wait to test :grin:

limpkin, 2023-08-29:
the recently released moolticute does have the feature, bundle v13 and the upcoming extension will implement the rest.
However i still prefer the "display totp after login" setting :)

My1, 2023-08-29:
cool, cant wait for v13

limpkin, 2023-09-01:
bundle v13 is available at https://beta-updates.themooltipass.com/ , extension to be released in a week or so :)

barathrm, 2023-12-28:
Finally tested this, works great :)

commit 282f10b3 (2023-08-28): get_totp fix, fix #372