mooltipass/minible #377: smc_decode_win.py script failing to decode 90% of passwords

LeaMaggot, 2023-02-05:
### Expected behavior
Using Python script to decode database

### Actual behavior
Script works as expected, no errors reported.
Logins are all decrypted but most of the decrypted passwords (90%) displayed like this
遙Ὦދ�닺鞬땣供헄牊歪⽰㨥ᣭਲ绾㋵洭ࣈ�쉃폍޽�◯忭펔乐ﳶꇟ�鴴ﶎ쥮颤

Others 10% passwords are correctly decrypted and displayed

i've done integrity check, reexported database with no success.

### Step by step guide to reproduce the problem

### Firmware Version
AUX MCU version:  0.73
Main MCU version:  0.84
Bundle version:  12

### Moolticute Version - If Involved

### Operating System
Mention if you are using either:
- Windows 7/10

  
Other operating systems are not supported.

### Mooltipass Extension
- If you're creating an issue to report a website incompatibility, please use the "Report incompatibilities with this website" item in the extension menu.
- If you're creating an issue to report a problem with our extension, please create another issue here: 
https://github.com/mooltipass/extension/issues

limpkin, 2023-02-05:
Hello! 
That's going to be a tough one to debug.... would you be able to give us more details about the ones that can be decoded and the ones that can't?
Mathieu

pcjc2, 2023-02-05:
Perhaps the OP could generate a non-secret test-case which reproduces the issue? (Perhaps it is UTF-8 or characterset related in some way).

LeaMaggot, 2023-02-10:
I don't see any difference between those who are correctly decrypted and those don't.
all of them are working correctly with mooltipass/moolticute
At the moment, to correct all the entries, i must manually edit all of them, suppress last pwd char, save then add suppressed char and save again.

limpkin, 2023-02-11:
nothing about special characters like àé^ ?
I'm not sure to follow you... are you trying to decrypt all of them in order to re-import them all at once?

LeaMaggot, 2023-02-16:
All the passwords are generated by the password generator included in your software or manmade.
some of them has special char like ^ ?

limpkin, 2023-02-16:
Let's try a different way: could you create a new user profile, with only a single credential, and let me know what passwords breaks the decryption process?

limpkin, 2023-03-12:
pinging @LeaMaggot