mooltipass/minible #379: Improving SSH key management

My1, 2023-02-17:
### Missing feature
1) There is literally no handling at all for keys with a passphrase, there's only a small info text at the import dialog, that only keys without passphrase are supported and when trying to import one with a passphrase, the importer just loops, doing nothing.
2) no Keygen, just imports

1) I would propose that not only there is SOME handling of keys with a passphrase (even if asking for the passphrase to decrypt might be too much, at least cancel with a message
2) similar how MP has a password generator, an SSH Keygen might be nice as the key would not be required to be unencrypted on the PC in the first place.

### Justification
make the SSH experience better in general

### Workarounds
make the key, unencrypted on the PC, which is less than ideal

limpkin, 2023-02-17:
I do agree that our SSH support is quite disappointing, which is why we usually push for FIDO2 for SSH login

My1, 2023-02-17:
which has its own problems, see #353 (although not at fault of MP), also does FIDO-SSH even work on windows e.g. with putty or what? like I know pageant and GPG-based things but not FIDO yet

limpkin, 2023-02-17:
there should be a few tutorials at the end of https://github.com/mooltipass/minible/wiki :)

schlomie, 2023-02-27:
I have been able to add password protected SSH keys to the minible.  No, you cannot do it through moolticute, but leveraging OpenSSH tools, it works just fine.   `ssh-add` prompts for the passphrase as expected, and is added to the list of keys the minible knows about.  This is persistent, so after the key(s) are added once, there is no need to add them again.

There _is_ some quirkiness though.   Some how the keys were added multiple times, but after a little cleanup in the SSH tab on moolticute, minible + mc-agent has been flawless.

My1, 2023-02-28:
openssh can connect directly to the MP or does that need the agent? when I try to enable the agent in MC it just gives an error about not packaged or whatever

schlomie, 2023-03-02:
The agent has to be running.  You'll need to install it:
```
$ go get github.com/raoulh/mc-agent@latest
```
`go/bin` will need to be in your `$PATH` so the binary, `mc-agent` will be available for moolticute.

What works best for me is overriding the location where the mc-agent socket gets created.  By default it creates it as a world-readable-writeable socket under `/tmp`.  To increase security, I use the `-a` switch on mc-agent to create it in a more secure location.  (Also a good idea to tighten your umask.)
```
Moolticute Settings
...
Moolticute SSH Arguments (Restart Needed):      [-a /more/secure/path/mc-agent.socket]
...
```

Anyway, once mc-agent starts up, and is writing to a known location, simply set the environment variable `SSH_AUTH_SOCK` to point to this location
```
export SSH_AUTH_SOCK=/more/secure/path/mc-agent.socket
```

So, when moolticute starts up, (if autostart is selected in settings) it will execute `mc-agent -a /more/secure/path/mc-agent.socket` which will fire up the agent in the background, creating the socket in this location.  Then, `ssh-add` will look for the agent identified by `$SSH_AUTH_SOCK` and work as it would with an OpenSSH agent running instead, including adding encrypted ssh keys.
```
# id_rsa or any ssh key...
$ ssh-add id_rsa
```
You'll only need to add the keys once.  The next time you ssh to a host, this key will be available via mc-agent.  minible will prompt you, asking you to export the ssh keys.  Click the wheel, and you're good to go.

Hope this helps.  Please let me know if I need to clarify anything.
`

t0mt3n, 2024-02-26:
I really want the feature of being able to create the SSH key on the device itself using the onboard RNG. This will make the key much more safe as it never leaves the trusted domain of the MiniBLE.