mooltipass/minible #381: Sync time over Bluetooth

ananthb, 2023-02-20:
### Missing feature

The mooltipass device should sync it's clock over Bluetooth in addition to USB.

### Justification

Tell us why you would like to see this feature added.

TOTPs are unusable from my phone only.
I need to have a PC and a phone to make this work.

### Workarounds

Are there any workarounds you currently have in place because the feature is missing?

I keep my device plugged in to a laptop near me if I know I'm going to need a TOTP.

limpkin, 2023-02-20:
oh that would be a neat feature for our android-companion indeed :)

My1, 2023-02-21:
honestly this only makes sense

wkennington, 2023-03-22:
This is not supported in thefirmware? Or this is just an app thing?

KDederichs, 2023-04-01:
@limpkin I'm looking into building that into the iOS App and got two questions:
1) What timezone? I'd assume UTC?
2) Does the device need to be unlocked for this? (If so why, I'd request that changed :D)

My1, 2023-04-01:
I'd guess UTC too as the MP doesnt display the local time and the primary point for the time is TOTP which works off of unixtime

limpkin, 2023-04-01:
1. UTC indeed, @deXol can yo uconfirm?
2. it does not :)

KDederichs, 2023-04-01:
Added it to testflight for the iOS app, so if you're iOS crowd feel free to check it out :)

deXol, 2023-04-01:
> 1. UTC indeed, @deXol can you confirm?
> 
>     2. it does not :)

Yes, it is UTC: https://github.com/mooltipass/moolticute/blob/master/src/MPDevice.cpp#L3901

My1, 2023-04-01:
cool. next step: the android app.

limpkin, 2023-04-01:
tracked here: https://github.com/mathfactory/mooltifill-android/issues/3