mooltipass/minible #382: FIDO Enhancement Collection

My1, 2023-02-21:
A mix of Bugs and Feature Reqs relevant to FIDO2 to keep everything together on a list

Bugs:
#380 - Bruteforce Protection problematic with allow/exclude lists of credentials for login and registering (ca. 3 sec per entry, seems to force fail after 5+ entries.)

Ideas:
#378 - set FIDO counter to always be zero for new setups to remove the need for time or other methods of keeping a counter
#304 - hmac-secret for Secret exchange (e.g. AAD or fido2luks)
#353 - CredProtect extension
Not yet Listed - non-resident Keys: removes the need to constantly sync Database between multiple MPs
Not yet Listed - CTAP2.1 (prerequisite to CredProtect)

Discussion/clarify:
#265 - my view so far is that as the MP currently does not use attestation certs, this is not yet relevant but will become important if/when attestation certs become a thing
#175 - explore the validity of silent Auth outside Webauthn. -> https://fidoalliance.org/specs/fido-v2.1-rd-20210309/fido-client-to-authenticator-protocol-v2.1-rd-20210309.html#op-getassn-step-sign mentions only that up=0 is invalid at the webauthn layer, however if 378 is implemented this will become moot.
#225 - Possibly add notes to manual or other documentation about skipping sending attestation

limpkin, 2023-02-21:
oh thanks for that :)

My1, 2023-02-21:
no problem. I think it's useful to have a quick list what's there to fix/add/discuss incl. quick explanations why