mooltipass/minible #393: Short password typed on credentials with TOTP when going back one step

kbembedded, 2023-03-21:
### Expected behavior
With all credentials, its possible to type a field, skip a field, and go back a field manually while using the miniBLE scrollwheel and display. It should not cause issues to type, skip, or go back in any amounts.

### Actual behavior
If a user uses the miniBLE to manually type a credential, type the name, password, go back and re-type password, the second time the password is typed is only a single character.

I've only observed this with credentials with TOTP and have been unable to reproduce anything similar with the username and password fields only.

### Step by step guide to reproduce the problem
- Open a notepad/word doc/go to website/etc.
- Unlock miniBLE
- Navigate to a credential with TOTP set up
- Use the minBLE to type the username 
- Use the miniBLE to type the password (observe its the full password)
- When prompted with TOTP, hold the scroll wheel to go back one step
- Use the miniBLE to type the password (observe it is only the first character of the password)

### Firmware Version
AUX MCU version:  0.73
Main MCU version:  .084
Bundle version:  12

### Moolticute Version - If Involved
N/A

### Operating System
- Windows 10

kbembedded, 2023-03-21:
In communicating with some folks in the IRC channel, and doing more testing, I've found that not _every_ credential I have with TOTP does this. But ones that reproduce it always do it, and ones that don't never do.

I need to go through all of my credentials and see what of them has TOTP and test all of them to see if I can narrow down a cause.

limpkin, 2024-01-06:
@kbembedded some update on this one? :)

kbembedded, 2024-01-07:
Its still a problem with some credentials, but, I was never able to find a correlation in testing.

The accounts I have that do it, if I copy their password to a test credential, the test credential does not reproduce it. So I'm not sure if its somehow related to the TOTP key used, something broken with my device's db, etv.

limpkin, 2024-01-07:
Could you recreate the credential altogether to try to see if it might have
been caused by a previous firmware version perhaps?

kbembedded, 2024-01-07:
I can but I'm not going to right now because I'd have to regen TOTP and re-store all of the backup keys, etc. But also, if somehow its related to TOTP, generating a new key may just happen to make it go away.

I'll try and re-create a credential in the next couple of days.

kbembedded, 2024-01-18:
I tried a few times, but was unable to cause it to occur with any new credentials that I created. Probably fine to just close this out.

closed 2024-01-20 by limpkin