mooltipass/minible #394: Support time entry on the device

wkennington, 2023-03-22:
### Missing feature
Current time entry for TOTP backing

### Justification
It would be nice for instances where you are without a computer but have a synchronized time source to be able to enter it if you really need access to your TOTP credentials. Obviously this is not the typical intended path but it would have been useful a number of times for me.

wkennington, 2023-03-22:
It will probably be a little annoying to deal with the year + month + day mapping in firmware. Care will need to be taken with the seconds + minute UI to allow seconds roll over during setting. A user should be able to get <5s precision in setting.

spoelstraethan, 2023-08-04:
The way the Precursor implemented this was pretty clever and simple, it takes the 2 digit year (solid because we'll all likely be using DNA authentication in 70+ years) , then the month (as 1-12) then day (as 1-31) then hour (as 0-23) and minute (as 0-59) and seconds (as 0-59). If you are entering the time manually you could just add 1 minute to the current time and wait until your clock/watch/phone rolls over before accepting the seconds as 00 or 01. I don't think <5s precision is impossible, though it would also be awesome to have an "edit time" that keeps the existing settings so you can zoom through to the minutes/seconds and just reset those if you have an issue with the TOTP/FIDO entries.