mooltipass/minible #395: smc_decode is missing a JSON file

talmage, 2023-03-28:
### Expected behavior
smc_decode prompts for my PIN, then decodes my smart card.

### Actual behavior
smc_decode asks for a JSON file.  It won't decode the smart card without one.  There is no documentation that explains the contents of the JSON file.

### Step by step guide to reproduce the problem

1. Run smc_decode.py
2. Observe the file chooser pop-up window.
3. ?

### Firmware Version
AUX MCU version:  0.73
Main MCU version:  0.84
Bundle version:  12

### Moolticute Version - If Involved

No

### Operating System

Pop! OS 22.04 LTS
  

### Mooltipass Extension

limpkin, 2023-03-28:
well, it asks for your database backup... how did you come across this script? :)

talmage, 2023-03-28:
It's part of the minible repo, right?  I learned about it while looking for a way to export saved login info to a CSV file.

I misunderstood the README.md.  I thought that it reads the saved login info from my Mooltipass BLE smartcard and then decodes it.  After a more careful reading of the very clear first sentence of README.md,

`This is a Python script that allows you to decrypt a Mooltipass backup by grabbing the AES key from your Mini BLE card using a standard USB smart card reader. `

I see that it's for decoding a backup.

I apologize for the false alarm.

closed 2023-03-28 by talmage