mooltipass/minible #398: PIN entry paranoid mode

Da-xy, 2023-04-20:
## Missing feature
I would like to have an option that allow me to display only the default asterisk instead of displaying the hexa number when entering the PIN code.
**Check Implementation proposal for more details.**

## Justification
I would like to entry my PIN code more safely in not trusted environments (public places, train or plane).
At the moment I cannot guaranty the fact that anyone did not see my PIN appear on the Mooltipass screen.
With the following configuration:
- Random String Pin : Enabled
- Display Pin on Press Back : Disabled
- Display Full Pin on Entry : Disabled

The fact that there is a significant time between the hexa number being displayed and the confirmation wheel click facilitate remembering the PIN code because in the end it is only 4 hexa number it is not hard to remember.
I learnt it the hard way with my coworkers doing jokes with my credentials... 
I could check behind my back every time I enter my PIN code but I would prefer another option.

## Workarounds
Lock me up alone in a room each time I enter my PIN code

## Implementation proposal
Counting the rounds wheel number or counting arrows animation on the Mooltipass will be the only way to enter the PIN code.
Obviously it is not compatible with Random String Pin option.

### 1st implementation : Strict mode
Do not give possibility to the user to display PIN code hexa numbers.
Identifying PIN code hexa value possibilities :
  - Counting the rounds wheel number
  - Counting arrows animation on the Mooltipass screen

#### Strict mode Pros/Cons :
- Pros:
  - Better security in untrusted environments
  - Fit better with typical password entry on a keyboard or cash machine PIN
- Cons:
  - Harder to enter PIN code
  - Higher risk to lock the Mooltipass card

### 2nd implementation : Back press capabilities only
Default is the asterisk displayed and the number will only be displayed using the back press feature. As soon as the wheel is turned again the number is replaced by the asterisk.
It is a mix using the already available option "Display Pin on Back Pressed"

#### Back press capabilities :
- Pros:
  - Better security in untrusted environments
  - Good tradeoff between usability and security in most cases the full PIN will never be displayed
  - Might be the most intuitive way
- Cons:
  - With this implementation it is not possible to see the latest digit

### 3rd implementation : Use wheel pressed capabilities
Default is the asterisk displayed but when the wheel is pressed simultaneously and scrolled up or down then it is possible to select the number you want to see. As soon as the wheel is released then the number is displayed and it is possible to modify it. As soon as the user move the wheel to modify the number then the number is replaced by the asterisk again.
It allows the user to modify the number of its choice without displaying the whole PIN just in case the user thinks an error was made during counting.

#### Use wheel pressed capabilities Pros/Cons :
- Pros:
  - Better security in untrusted environments
  - Good tradeoff between usability and security in most cases the full PIN will never be displayed
  - Give better usability in general for PIN entry because if you need to go back at the beginning of the PIN you need to do 3 press back which can be long
- Cons
  - Do not know how it will behave with the back press feature. I think the pressed scrolled wheel is already used in login section to go faster on domain selections. I do not know if it is counter intuitive or not.

In my opinion the third implementation might the best as it it will increase the security level and PIN entry usability.
It might be two features in one on this particular implementation.

I look forward to see your replies on this topic.

Anyway many thanks to the Mooltipass team for this awesome product ! <3

Da-xy, 2023-04-25:
After all I tried writing something about this feature request.

It is none of the 3 implementations I propose it is a 4th one because implementing the 3rd implementation was a little bit complicated because there was no wheel "release" action that directly do the job.
I only found low level functions and I think it would have been to complicated.

So basically the difference is just that by default when the feature is enabled only asterisks are displayed and you can scroll up or down to change the digits.
If you get lost in your count you can use the wheel press + up or down to display the digits.
Here is an example in video.
It is recorded with the emulator so you will not be able to see the difference between wheel scroll and wheel press scroll but it will give you an idea.

https://user-images.githubusercontent.com/7549969/234258587-1dd70be3-c9f2-4f65-b9f1-7581fe2acf1e.mp4

I would like to go further with the following things but I do not know how to start :
- Test it on my Mooltipass
  - Generate a new bundle
  - Upload it on my mooltipass (maybe not recommended? I do not know the risks...)
  - Manage the firmware upgrade
- Develop the feature on moolticute to be able to enable the feature as it is disabled by default
  - I set up the environment with QT creator but unfortunately I am not able to debug the application as after the main thread is launched. The break points are not triggered...

Anyway if you want to test the feature on the emulator here is the branch : https://github.com/Da-xy/minible/tree/pin_entry_paranoid_feature
I do not know if I should create a PR or not ?
If you want to enable the feature you need to change the following value by `TRUE` :
[image]
It requires to create a new `dbflash.bin` and `eeprom.bin` because it needs to be created once.

I think I miss a lot of things but I wanted to give it a try! Maybe there is a mailing list or chat group where I can find those answers?

Regards!

limpkin, 2023-04-25:
wow... that's really really neat!
Regarding qt creator, IIRC you need to start the daemon, then the GUI but when the daemon is automatically killed you need to manually restart it. @deXol is the person to ask :)
You can definitely create a PR that I can review! If you want to talk to me or the team, we're on IRC on #mooltipass on irc.libera.chat

Amazing work, congrats!

Da-xy, 2023-04-26:
Thank you very much!

I joined the IRC chat I will reach out when I got a little bit of time.
Okay I'll try this and if I get some trouble I will reach this person !
I created the PR ! :)

CGuy-1, 2023-05-29:
I would like to see option 5)
Display the numbers AS the wheel is being scrolled and an asterisk when the wheel stops moving (after 250 ms settable by the user) and moves to the next after the normal amount of time.

Alternatively, option 4) but instead of displaying asterisks, display unique random symbols. 
For example each scroll could display !@#$%^&()-+={}<> so you know the value changed. The order would be random for each time you move to a new column.  For the random starting number, you can display it then just the symbols in a random order.

limpkin, 2023-05-30:
@CGuy-1 you'll get to test @Da-xy implementation with bundle v13 :)
@Da-xy should we close that issue?

Da-xy, 2023-05-31:
> I would like to see option 5) Display the numbers AS the wheel is being scrolled and an asterisk when the wheel stops moving (after 250 ms settable by the user) and moves to the next after the [...]

Hi @CGuy-1 !

Thank you for the interest regarding this feature request.
For the 4th implementation, I am not sure to understand correctly as even if the number is hidden (with the asterisk) when the wheel is released then it will still be possible for someone to guess the correct number right ?
Regarding the 5th proposal I find it interesting, it might be a good idea to try to implement something around it later if what I implemented does not fit.

@limpkin I think we can. :)
We will open it again if needed.

closed 2023-05-31 by limpkin

limpkin, 2023-09-01:
bundle v13 is available at https://beta-updates.themooltipass.com/ !