mooltipass/minible #400: Passkeys support

ai212983, 2023-05-03:
Hello guys.

Tech industry seems to move from passwords to passkeys. 
Is MiniBLE going to support passkeys at some point in the future? 

Thanks.

limpkin, 2023-05-03:
Hello!
The same was said about FIDO2, which is why we implemented support for it :).
To answer your question, we definitely want to support passskeys :)

VincentVanlaer, 2023-05-05:
It seems like passkeys is more or less a rebrand of the existing FIDO2 standard, see https://www.yubico.com/blog/a-yubico-faq-about-passkeys/. In any case, I managed to use my mooltipass to login to my google account using passkeys.

My1, 2023-05-08:
yeah Mooltipass is in someway a subset and also in some way an extension of the FIDO2 standard.

1) no or self attestation is explicitly part of passkey stuff, and not really optional, unlike before (severely benefits the MP by the way)
2) Passkeys are (currently) only done in a resident manner (the MP luckily doesnt suffer from the same SEVERE lack of credential storage the Yubico keys do, I mean only 25 lol?)
3) Passkeys are supposed to be syncable (usually across the vendor's cloud), so MP doesnt need to care.
4) Passkeys always go with UV, not sure if it's the spec or how platform FIDO always behaves, but the MP uses the smartcard unlocked state as UV anyway
5) Passkeys are intended for Passwordless login (that's awesome, saves space on the MP)
6) Passkeys are usually platform-based, so site support is the biggest issue on how they handle it and if they send in the attachment parameter and how much of a pain they prove to be.

ai212983, 2023-05-12:
@VincentVanlaer Oh, awesome, can you please describe the process for dummies (like me) how to login to google account with Mooltipass and passkeys?

My1, 2023-05-12:
the most important thing people might not know is if they used security keys on that account already, to make sure to have less than 5 prior to adding the MP, otherwise delete and re-add them.
well for starters make sure you have less than 5 security Keys active on your Google Account.

limpkin, 2023-08-30:
should we close this issue as the Mini BLE is passkeys compatible? I've updated the main website to explicitly state that.

Tsaukpaetra, 2023-08-31:
It's weird, I thought I had it working but even though I registered mine to Google, actually trying it results in a fail with "We don't recognize this key" message.

CGuy-1, 2023-10-12:
>  I thought I had it working but even though I registered mine to Google, actually trying it results in a fail with "We don't recognize this key" message.

That seems to be an error with Google. Does it work with a Yubikey? 

I know Google wants everyone to use their authenticator app as it collects information on the sites you use and add this data to all the other data they've collected on you.

My1, 2023-10-12:
their authenticator app isnt even in any way related to passkeys. that thing is just TOTP, iirc it should work by now, (unless you have like 18 other Security key or passkeys in your account lol)

as mentioned the issue was that the MP uses self attestation which is a comparatively new mode.

at worst if chrome asks whether to send attestation or not just tell it to skip and it should definitiely work.

CGuy-1, 2023-10-13:
Yes, at this time they just do TOTP and push notifications, which is really insecure. Google, Microsoft, et al. have all committed to implementing the passkey.

The problem I have with these S/W authenticators is the extra information they harvest while providing the service. It is for this reason that I wish to stay with a H/W solution like MP and Yubico.

Tsaukpaetra, 2023-10-13:
> That seems to be an error with Google. Does it work with a Yubikey?

I don't have any other FIDO devices. 😅 The TOTP works fine after the MP syncs time, but when trying the passkey method the MP doesn't respond at all to the request (if it's even registering it? Maybe a UI improvement). At present I only have two passkeys in the FIDO section, Microsoft and Google (the MS one was added first), and the Microsoft one works fine, prompts and all.

My guess is that it's not prompting for a specific key and the MP is just "guessing" and sending the first one on the list and Windows is just bleching at it since it doesn't match? Who knows, I don't have my system set up for debugging this. 😅

alanrick, 2023-11-14:
> should we close this issue as the Mini BLE is passkeys compatible? I've updated the main website to explicitly state that.

I use passkeys on my MiniBLE and love it. Also that fact that more and more platforms support it (e.g. just now to logon to Github effortlessly).

But I don't think we should close the issue until the Mini BLE supports FIDO2 over bluetooth. After all, bluetooth is the signature distinguishing feature of this device.

EXTgithub99cd2, 2023-11-17:
I am confused. 
Based on this issue the main website now mentions "passkey" support. But I also read some yes/no/maybe as in terms of working implementation.

My knowledge of the difficulties is limited, making it impossible to assess if this mooltipass is a futureproof acquisition as passkey device. 

Can the community elaborate on the level of support as passkey device on a user level? Does the implementation match with industry standards, tech giants default implementations, current developments and forward looking insights? 

It is the main selling argument at this moment going forward I would guess.

My1, 2023-11-17:
There are really only 3 bigger "issues" in the Passkey front:
1) self-attestation (some sites cant really deal with it, as they don't understand it especially those who were FIDO-capable for a while already, some browsers allow you to just strip off the attestation, which can help)
2) no support for e.g. HMAC-secret, which mainly affects Microsoft, where you cannot use a Mooltipass yet.
3) due to the fact the Implementation the MP uses still uses a counter (and it cant get away with it for existing credentials anyway) the Passkeys are currently reliant on the MP's ability to keep and sync time (like a PC with Moolticute every once in a while or after a full reboot). Apparently modern multi-device Passkey implementations such as those used by Smartphones just keep the counter at zero which is according to spec. (dunno when that was added tho)