mooltipass/minible #401: Add debounce to push button

lwYeo, 2023-05-07:
### Missing feature
Debouncing does not seem to be present when pressing down the button.
Maybe adding around 100ms would be nice.

### Justification
Sometimes the button "double-clicked" by itself, occurrence seem to be 1 in 10 or less (at least on my device).
Although it does not seem much, but it is rather annoying when it happens, especially when entering the last digit pin, causing unwanted "invalid pin" incidents.

limpkin, 2023-05-07:
there's actually already a debounce: https://github.com/mooltipass/minible/blob/master/source_code/main_mcu/src/INPUTS/inputs.c#L275, set to 100ms

lwYeo, 2023-05-07:
> there's actually already a debounce: https://github.com/mooltipass/minible/blob/master/source_code/main_mcu/src/INPUTS/inputs.c#L275, set to 100ms

Nice.
Since different unit may have different tolerances, would it be better if we can customise it?
If not can we increase to maybe 150ms-200ms, to reduce this issue further?

limpkin, 2023-05-07:
that's indeed a great idea!

My1, 2023-05-08:
might this be the same reason why i sometimes dont get a wheel move when moving quickly through the numbers when trying to enter my pin?

limpkin, 2023-05-09:
not exactly no

CGuy-1, 2023-05-29:
I have this issue with my early unit, it seems the scroll wheel starts to do this after lots of use. This will just get worse and worse until the wheel stops working. I've replaced this unit with a new one that I think has a better scroll wheel installed. The new one doesn't exhibit any issues but it's only been 6 months.

seviu, 2023-05-31:
I got same problem

limpkin, 2023-05-31:
@seviu : can you expand on your symptoms? :)

limpkin, 2023-08-30:
@CGuy-1 is this still happening? I could offer a device swap...

CGuy-1, 2023-09-06:
Yes, I suspect the scroll wheel contacts are broken or worn out. This is an early unit and I use it all the time so it may just need a new scroll wheel component. The battery doesn't hold a charge any more (and yes I've run the refresh several times), is it possible to replace it?

limpkin, 2023-09-06:
send me a quick email, we'll work something out :)