mooltipass/minible #402: Throw out gyro (both when no card is inserted and as an option in general)

My1, 2023-05-08:
### Missing feature
I noticed the gyro wakeup having been active while there wasnt even a card in the MP, and thought especially while it's being carried around or whatever that aside from actually waking up the device just keeping an eye on the gyro might cause some unneeded battery drain, and especially while there is no card in the MP there is literally nothing the user can do, except for putting in a card (or perhaps fully turning off the MP, which isnt a great idea anyway due to the loss of time.

so I would propose to
1) just throw out the gyro while no card is in
2) give an option for gyro stuff that could be a dropdown with 3 options:
a) active
b) disable on battery
c) always off

### Justification
Tell us why you would like to see this feature added.

less battery drain is always great

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

currently not really.

limpkin, 2023-05-09:
The gyro wakeup only happens when the device is battery powered, I'm therefore guessing your device wakes up because the wheel gets pushed?

My1, 2023-05-11:
I was mostly thinking of making this short suggestion (where you said good suggestion) into an issue, if the fw already does stuff to work around this cool, and can be closed.

https://matrix.to/#/!ruKpkRrOFphGsnWhet:matrix.org/$8XJTf88qBz7XSDxS3-kqKPYh8nOikfO8oX5jqanCzJE?via=libera.chat&via=matrix.org&via=van-hemmen.com

limpkin, 2023-05-11:
It seems I was mistaken when I wrote that reply on matrix :(

limpkin, 2023-08-30:
I'm closing this issue as the gyro isn't used when the device is on battery

closed 2023-08-30 by limpkin