mooltipass/minible #403: Ability to set minible clock/time manually using the minible's display and scroll wheel

barathrm, 2023-05-19:
### Missing feature

### Justification

Sometimes I need to generate an OTP code while traveling or when not having moolticute or maybe even mooltifill available to set the time on the minible. But, I probably have an accurate clock available on my phone or the device where I need to input the OTP code.

I wish I could manually set the time on the minible using the minible's display and scroll wheel. 

### Workarounds

There are no work-arounds. If the time on the minible is not set and I don't have moolticute or mooltifill available, I am unable to generate OTP codes using the minible.

spoelstraethan, 2023-08-04:
This is a duplicate of #394 I believe.

limpkin, 2023-08-05:
good point! thanks @spoelstraethan !

closed 2023-08-05 by limpkin