mooltipass/minible #404: Mini BLE does not scroll backwards past the start.

CGuy-1, 2023-05-30:
### Expected behavior
When selecting a credential to login manually, the device doesn't scroll backwards past the start of the DB. So for Yahoo.com, you scroll backwards to find it quickly.
### Actual behavior
The scroll stops at the start of the DB list going backwards but will scroll from the end to the start going forwards. If Yahoo.com is the last entry you are forced to scroll through all entries, if you go past you're back at the start and need to go through the list again.
### Step by step guide to reproduce the problem
Select Login, move scroll wheel to descend the list, it stops at the first entry.
### Firmware Version
AUX MCU version:  0.73
Main MCU version:  0.84
Bundle version:  12

### Moolticute Version - If Involved
N/A (v1.01.0)
### Operating System
- Windows 11

### Mooltipass Extension
N/A

limpkin, 2023-05-30:
that's odd... could you perform an integrity check?

CGuy-1, 2023-05-30:
How do I perform the integrity check?

limpkin, 2023-05-30:
dedicated button in the synchronization tab :)

CGuy-1, 2023-05-30:
Integrity check came good.
[image]

limpkin, 2023-05-30:
can you try again now?

CGuy-1, 2023-05-30:
Same, it won't go past the beginning of the list in reverse.

limpkin, 2023-05-30:
in memory management mode, are you able to see passwords at the end of the credential list?

CGuy-1, 2023-05-30:
Not at the end of the list. I need to select and press the icon in the password field and approve on the device to see the password.

I have noticed that for some accounts I never get the prompt on the device to see the password so it comes back denied. Other accounts I'm able to get the password as expected.

CGuy-1, 2023-05-30:
The first one I found that failed was the 2nd credential of the service  **a8.com**, the first credential works correctly.
The next service, **b1.com**, has one credential and it also failed but the single credential for **a7.com** worked.
I went through the list and the 2nd credential of **m3.com** failed, the first worked correctly. The previous service, **m7.com**, has 4 credentials that all worked.

After correcting the two errors in the database, I get the prompt for every credential I've tested and the scrolling backwards past the first entry is working.

CGuy-1, 2023-05-30:
As an aside, possibly related, I originally had two Notes, now when I check there are 6, 3 copies of each one. Is this another corruption? Do I need to manually backup the information, delete all the notes and manually recreate them?

limpkin, 2023-05-31:
unlikely, but there was a bug quite a long time ago on moolticute that could possible duplicate notes... can you confirm you're running the latest moolticute?

closed 2023-05-31 by limpkin

CGuy-1, 2023-05-31:
Moolticute v.1.01.0, Check for updates returns that I'm running the latest.
So I can just delete the extra notes?

What caused the corrupt DB that stopped the reverse scrolling and password display? Is there something I need to do to avoid it in the future?

limpkin, 2023-05-31:
you may indeed... just check that the others can be read.
Your bug won't happen in the future as we fixed the underlying issue on moolticute :)