mooltipass/minible #406: Update firmaware from 5 to 12 fails

paweljasinski, 2023-08-14:
### Expected behavior
Update completes w/o error

### Actual behavior
After selecting a bundle file, msg box displays: "Upload bundle finished with error."

### Step by step guide to reproduce the problem
1. Download bundle using link provided in the multicute app , from version 5 to version 12
using the link https://www.themooltipass.com/updates/index.php?sn=2771&bundlev=5
```
[rejap@zet Downloads]$ md5sum update_2771_from_bundle_5_to_12_25a3d547a73c005b7980a03ab019df84.img
313ace97c4b130a8fb6ff0956bb2a4f5  update_2771_from_bundle_5_to_12_25a3d547a73c005b7980a03ab019df84.img
```

2. Follow instructions from https://mooltipass.github.io/minible/updating_your_device
2.1 Reconnect usb
2.2 Remove card, insert reversed
2.3 Shift-F3, select BLE-Dev tab
2.4 select file from step 1.

### Firmware Version
AUX MCU version:  0.66
Main MCU version:  0.77
Bundle version:  5

### Moolticute Version - If Involved
1.01 Linux Arch, from AUR

### Operating System
Linux Arch x64, up to date

### Mooltipass Extension
no extension involved

limpkin, 2023-08-14:
that may be a linux related issue.... can you try on a windows OS? I remember we had a similar issue in the past.

paweljasinski, 2023-08-15:
yep, using windows is a valid workaround.
However, I did not manage to get it working with windows running as vmware guest. I had to use windows on the bare metal (have one laptop for this special occasions :-)