mooltipass/minible #407: If only TOTP is saved, it is not shown

EugeneRymarev, 2023-09-28:
### Expected behavior
If the device is connected via BLE or USB to a controlled device, then make an input request and enter the current TOTP.

If the device is not connected, then report a connection error, since entering TOTP requires knowing the current time.
### Actual behavior
Blank screen after asking "Display Credential?" and a positive answer to it.
### Step by step guide to reproduce the problem
Save your credentials with empty login and password fields, and then add the TOTP code.
### Firmware Version
- AUX MCU version:  0.73
- Main MCU version:  0.84
- Bundle version:  12
### Moolticute Version - If Involved
- v1.02.0
### Operating System
- Windows 11

CGuy-1, 2023-10-12:
> Save your credentials with empty login and password fields, and then add the TOTP code.

Interesting, what is the use case for this? Why are you needing to enter a TOTP without a username/password? I've only ever known TOTP to be used as a second factor when entering username/password credentials.

I have seen that when Moolticute enters credentials for me that it only displays the TOTP where as if I use the Mooltipass to enter my credentials, it will send them instead of just displaying them (which I prefer).

EugeneRymarev, 2023-10-13:
Hello, @CGuy-1!
There are sites that do not have classic login-password authorization (it is replaced, for example, by Google authorization), but they do have 2FA. Therefore, such functionality is necessary.

limpkin, 2023-10-14:
I'm impressed you managed to setup such a credential as I don't seem to be able to enter an empty credential to the device **and** setup a TOTP on a credential that doesn't have a login...  it seems the receipe was add a credential with a login and totp in MMM, deny the password prompt, then delete the login... but in that case I do get prompted to type the totp!

could you therefore write me a more detailed step by step guide, using the latest 1.02.18 at https://betas.themooltipass.com/v1.02.18-testing/ and bundle v13 from beta-updates.themooltipass.com

EugeneRymarev, 2023-10-15:
@limpkin, I don’t remember how I did it, but I suspect that I added the service login and password at the beginning, and then cleared the login and password fields.

I installed the new firmware and the new Moolticute app.
The Display Credential window is still empty.
By the way, there is no way to view the code for 2FA. Login and password are possible, but 2FA code is not. It might be worth adding this feature. Or at least add a visualization that 2FA is installed.

limpkin, 2023-10-15:
from this screen capture, no TOTP is set for that credential though (when there's one, you get a delete & edit button).
for the empty prompt, is it possible that the device showing an empty password but a login made of a single space character?

EugeneRymarev, 2023-10-15:
@limpkin, nope. Login is empty. No spaces. Yes, TOTP is empty too. HMMMM.

limpkin, 2023-10-15:
I've created this issue then: https://github.com/mooltipass/minible/issues/412
However, given that I can't recreate yours... I'll likely have to close this one unless you can give me a list of steps to reproduce this issue...

closed 2023-10-16 by EugeneRymarev