mooltipass/minible #408: Automatically lock on bluetooth disconnection

hoijnet, 2023-10-04:
### Missing feature
Add an option to automatically lock the device upon bluetooth disconnection (if the device is connected to a device).

### Justification
I tend to have my Mooltipass connected to my computer by USB and it's not unreasonable to expect that users sometimes forget to bring the device or the smart with them when they lock their screen.

It would be amazing to use the bluetooth connectivity to automatically lock the device when disconnected due to out of range.

This would be a safety measure as a last resort, but to ensure the device is not left logged in by mistake, or when the USB disconnection does not trigger locking which happens sometimes.

### Workarounds
Bring the smartcard with me.

hoijnet, 2023-10-13:
Smart! Of course, even better. And I agree the smartcard should not be left with the device unattended, for many reasons.

As the pin is displayed on screen, it should be noted as a risk that is can be seen by others. The real security is is in the smartcard that should be carried!!

As this solves the issue, I'll close this one!

closed 2023-10-13 by hoijnet

CGuy-1, 2023-10-13:
Yes viewing the PIN is a known security risk which is why you might want to enable **Random Starting PIN**. This will prevent a user from counting the wheel turns to deduce the PIN number. It is necessary to shield the display from prying eyes as you enter it.