mooltipass/minible #411: Add swiss german keyboard

limpkin, 2023-10-14:
### Missing feature
swiss german keyboard layout

### Justification
fairly obvious

### Workarounds
none

My1, 2024-07-29:
stupid idea, make a few (e.g. 3 or so) slots for custom kb layouts. (especially useful for slightly more unusual layouts like "German (no dead keys)"

Sazoji, 2024-07-29:
If you could load like how qmk references custom keymaps to translate characters to key codes that solves half the issue,
IDK how dead keys should be included or what special characters require a custom keyboard translation input (as someone with the same exact keyboard at home and at work I don't get the pleasure of dealing with language-specific keyboards) but if you're only translating the changes to include the language-specific character-> kbd output seems very short, include all the üüüs you want.
https://github.com/qmk/qmk_firmware/blob/master/quantum/keymap_extras/keymap_swiss_de.h

limpkin, 2024-07-29:
at some point it did cross my mind to allow upload of custom key maps. We never got to it because of the amount of time needed to implement it and the very few users who could benefit from it :/