mooltipass/minible #413: Entering recorded data regardless of system layout

EugeneRymarev, 2023-10-17:
### Expected behavior
Entering recorded data regardless of system layout. If the device contains the entry `myemail@mail.com`, then enter it that way, even if the current system layout does not match.
### Actual behavior
Instead of the expected `myemail@mail.com`, `ьнуьфшд"ьфшдюсщь` is entered on the RU layout.
### Step by step guide to reproduce the problem
1. Create a record `myemail@mail.com`
2. Set the system to the RU layout or any other that will not coincide with the layout in which the `myemail@mail.com` entry was entered
3. Change the current layout to RU (or another selected one)
4. Manually select this entry on Mooltipass and ask it to enter the data into the controlled device
### Firmware Version
AUX MCU version:  0.74
Main MCU version:  0.86
Bundle version:  13 (beta?)

### Moolticute Version - If Involved

### Operating System
- Windows 11
- Android 14
### Comment
I still can't understand one thing. Why is it that when a device automatically enters a login and password, it enters exactly as it is written in its memory, regardless of the current layout of the controlled device (be it a PC or a phone); and when I manually select a service and ask the device to enter a login and password, does it enter in the current layout? For example, my device enters the email `myemail@mail.com` as `ьнуьфшд"ьфшдюсщь` - that is, in the current layout of the managed device RU.

limpkin, 2023-10-17:
> the entry ***@***.***, then enter it that way, even if the current
> Instead of the expected ***@***.***, ьнуьфшд"ьфшдюсщь is entered on
>    with the layout in which the ***@***.*** entry was entered
> layout? For example, my device enters the email ***@***.*** as

We are using two different mechanisms to send credentials to the computer.
When using the extension, we are using a custom HID channel while when
recalling credentials through the device we are using a "keyboard HID"
channel.
The latter channel uses key positions and not characters (that's the HID
standard) so the mini ble needs to be aware of the keyboard layout selected
on the computer.

There's therefore no solution for your issue...

EugeneRymarev, 2023-10-17:
@limpkin, Why can't change the manual input mechanism to imitate automatic input?

limpkin, 2023-10-17:
For the same reason you need a keyboard to type things in your computer..,.

EugeneRymarev, 2023-10-17:
@limpkin, I don't understand. What's the problem with using custom HID for manual entry? 🤔

limpkin, 2023-10-17:
because your program (let's say, notepad) only "listens" to keyboard presses

EugeneRymarev, 2023-10-17:
@limpkin, Fine. Let me pose the question a little differently.

Is such an implementation impossible or possible, but requires a lot of labor?

Something tells me that there are no unsolvable problems.

limpkin, 2023-10-17:
I don't think it's actually doable to have a software simulate key presses
into any open program... but I may be wrong

EugeneRymarev, 2023-10-17:
@limpkin, check Clickermann. That tool from russian developer.
[Direct download link](http://crapware.aidf.org/files/9/)
[Main page](http://crapware.aidf.org/page/clickermann)

limpkin, 2023-10-17:
I'm sorry, I don't speak russian.
I'm going to close this issue for now as implementing the feature you're requesting requires an external program that could directly requests credentials to the mini BLE.

closed 2023-10-17 by limpkin

EugeneRymarev, 2023-10-17:
@limpkin, The program includes English. The site can be translated using the built-in translator in the browser. In any case, this is just an example that this is possible.

I can (if I read the documentation) write a simple Python script that will enter any information in any language into any window.

Why need new external program if you already have Moolticute?

limpkin, 2023-10-17:
right... it sounds like a feature request for moolticute then :)