mooltipass/minible #414: More faster power off

EugeneRymarev, 2023-11-09:
### Missing feature
Simply turn off the device from the main menu without confomation, without having to hold the wheel button twice and confomation.

### Justification
This would be faster and more convenient (for some users).

### Workarounds
Hold wheel button twice and confirming.

My1, 2024-06-16:
stupid idea just power down everything except the RTC once you pull the card (not that you can do anything without a card inserted, sure it's a nimh battery instead of a non-rechargeable button cell but it surely should last longer than like a week

limpkin, 2024-06-17:
I'm actually doing this very thing for the next gen device :)

Chris-70, 2024-11-12:
If there is just a power off menu selection that is confirmed with the standard button that would be an improvement.
Also a longer press to turn on as it turns back on when I have it in my pocket and the button get bumped.